jeffeb3/sandify #6: Get Turtle Working

jeffeb3, 2017-07-02:
My intention was to bring in as much of this as possible:
https://github.com/wrschneider/js-turtle/blob/master/turtle.js

But it's just very intertwined with the image canvas stuff, which seems (to me) to be completely incompatible with react-bootstrap and react.js. I found an example of a react canvas, and it's drawing a rectangle. Maybe it would be better to either: 
a) copy-paste the turtle.js into TurtleCanvas.js and start hacking away at getting the functions to use the react canvas instead or
b) Use turtle to generate a list of positions, x, y and then have TurtleCanvas be sent in that list of locations or
c) ditch the turtle alltogether, if it's not helping us.

I'm inclined to do b) but Maybe @jamesd256 has a better way of doing that? Maybe I'm thinking about this all wrong?

jamesd256, 2017-07-05:
Could you provide a link to the example of react canvas you found?  Was it [this one?](https://github.com/Flipboard/react-canvas).  I found some interesting information about the subject [here](https://blog.lavrton.com/using-react-with-html5-canvas-871d07d8d753).  There is a react-canvas out there which is not really an HTML 5 canvas, so might be misleading.

Can you help me with a basic understanding of why react is incompatible with HTML5 canvas?  I notice in your TurtleCanvas.js, you use the following to obtain the context:

 var context = ReactDOM.findDOMNode(this).getContext('2d');

Is your concern about getting access to the context?  While looking into how to do this I came across this post, which offers a couple of different ways to obtain canvas context access.  

It seems to me that react + canvas should function fine, but maybe you feel it's not the correct approach for some reason.  The turtle_port branch contains turtle.js ported to be a react component, and this can be adapted to output either gcode or raw vertices.

jeffeb3, 2017-07-05:
Sorry, this issue is out of date. I think it should probably be closed.

The example I found was this: http://jsfiddle.net/JMZc5/1/ but it didn't work until I did the ReactDOM.findDOMNode call.

I'm currently happy enough with the drawing that we have in MachinePreview.js. If using 'react-canvas' is more right/better/more flexible, then I'm fine with that too.

I'll open another issue where we can integrate turtle.

closed 2017-07-05 by jeffeb3