jeffeb3/sandify #17: Machine Limits

jeffeb3, 2017-07-05:
Recompute vertices so that the shape is as preserved as possible when the vertices exit the prescribed machine size.

jeffeb3, 2017-07-07:
Works great, in the scenarios I've used.

jeffeb3, 2017-07-07:
I broke it... It works well for big squares, but not for stars, anymore.

GhostDaddio, 2017-07-15:
I've been working on the machine limits and I think I've got a mostly brute force method.  It mostly works, but when I tried to push it, I was denied access.   How do I do that?  The LimitEnforcer.js file I tried to push is attached.
[LimitEnforcer.zip](https://github.com/jeffeb3/sandify/files/1149996/LimitEnforcer.zip)

jeffeb3, 2017-07-15:
Thanks. I've been cooking up some progress offline about this, but I appreciate some help. You can't push to this repo (I think you can't push to any branches either). I will look at the zip, but in the future, you'll need to learn how to make a Pull Request. If you're not on the project, then you'll need to fork the project first:

https://help.github.com/articles/creating-a-pull-request-from-a-fork/

BTW, are you someone we know on the forums with a different name?

jeffeb3, 2017-07-15:
[image]

Not a big fan of comments, eh? I think I am one too many beers away from understanding this without some direction, or variable names. I did run it against my first diabolical case (above).

jeffeb3, 2017-07-15:
I'll push my working branch so you can see what I've been doing. I am segmenting the problem into which of 9 boxes the vertex and the previous vertex lie, because the hard cases are when the vertices go out from one end and enter from another. I think I'm pretty close, I just need to concentrate for a moment on it.

jeffeb3, 2017-07-15:
I pushed my stuff to limits_2. I usually do a little cleanup when I'm done, and I'm not done yet. I think I only have trouble with the give up section, IIRC, and I was thinking I might take those cases and bisect the lines into three vertices, and test those two lines until I don't have to give up.

GhostDaddio, 2017-07-15:
I'm a big fan of comments, unfortunately, I'm not so good at them.  As for the other forums, I go by BT there.  And, as I'm sure you've guessed, the whole mechanics of github are foreign to me.

I must not understand the geometry of canvas, etc.  When enforceLimits is called, it is called with the width&height divided by two of the "printable" area.  I assumed the drawing was centered on the canvas and therefore, the extent of the drawable area was from -size_x to +size_x and -size_y to +size_y. where size_x/size_y are adjusted to be half of the range from min_? to max_?.  So, I'm missing something on how adjusting the minimum from 0 (I think maximum worked) affects how I was calculating the extent.

Separately, I tried to test limits_2 branch and the file "Victor" is missing from github.

One more thought to make sure we are somewhat synced.  When the drawing exceeds a limit, are you planning on the line following along the outer edge until the drawing returns to within bounds?

jeffeb3, 2017-07-15:
OK BT/GhostDaddio, that will help me keep my head on straight. Git is complicated if you haven't used it, and github is another layer that adds quite a bit too. They really are the best way to collaborate with strangers though.

Those limit assumptions are correct, at least that's what I thought it was doing :) .

The vertices are centered at 0,0 and the canvas drawing then converts those to center that in the middle of the absolute area of the canvas. So 0,0 vertex in the limits code ends up being drawn at 250,250 or so in the canvas.

Victor is an external dependency, so you have to do `npm install` to update your local packages.

When the vertices leave the -size/size boundary, I want the limited vertices to follow the same path while inside the boundaries (so no changing of the shape) and they travel along the boundary when the original is travelling outside the boundary.

[image]

The star case is the easy one, it's when the square has a whole loop's worth of vertices that are outside is when you get the problems,

GhostDaddio, 2017-07-15:
My attempt, in short, was to find out if the starting point or ending points were outside of the bounds.  If so, then iterate along the line between the two points to find the first point and last point that are within bounds.  It's brute force and no elegance.  Using your neighbor area approach, I'm sure there's a more elegant method.  Something like if it is in one 0b1000, 0b0001, 0b0100, 0b0010, then check the line for intersecting with the nearest edge (that's what my GetXValue and GetYValue functions were for - input 2 points on a line and provide the X or Y value and find the corresponding value).  For the corner cases, need to check if the line intersects with the nearest two edges.  Since we are dealing with lines, there should be a simple comparison check to make sure that the intersection point is in the direction to the next point.

jeffeb3, 2017-07-18:
I hope I'm not steamrolling you GD. I did have my solution almost finished, and it's doing well in most cases. I do appreciate the help. I have a PR #37 which I think you've looked at. I think I've fixed the last remaining case, and I can't seem to break it. I do notice some slow down with the vicious logo, when I make a lot of loops that are out of bounds. Other than that, and the Victor dependency, I think it's pretty good.

jeffeb3, 2017-07-18:
[image]

GhostDaddio, 2017-07-19:
I like it.  The star and vicious1 work great,  The triangle and the square seem to still have problems.  Here's what I see when I select the basic square, spin and rotate and then increase the square size.

[image]

I don't feel steamrolled at all.  I've been busy, too busy in fact to do much other than this simple test.  I might get some time this weekend to play some more.

GhostDaddio, 2017-07-19:
Partial nevermind.  I pulled the limits_2 again and re-tried it.  Now the square seems to work, but the triangle doesn't.

[image]

jeffeb3, 2017-07-19:
I can confirm that triangle is in the latest code. Darn. I thought I had it. I'll look at that soon.

jeffeb3, 2017-07-19:
[image]

jeffeb3, 2017-07-19:
It was coming back through the area, from the top, and leaving out the side. Then the next vertex was along the bottom, so it gutted across the scene again. I added another nearestVertex(lin_end) in that case, so it would go all the way to the bottom in that case. Thanks for testing this.

GhostDaddio, 2017-07-19:
A quick check and it still works for square, vicious1 and star in addition to working for triangle.  Well done!

closed 2018-11-16 by jeffeb3