jeffeb3/sandify #18: Base Input component?

jamesd256, 2017-07-05:
See commit c7ffa3a.

jeffeb3, 2017-07-06:
If this was C++, I would be a big fan of making a base class, to define the interface for the children, and to have a location for common functionality. In Javascript, I guess it shouldn't be any different, but I got confused about a few things when I tried it. In general, I'm a fan of this architecture.

jamesd256, 2017-07-06:
I'm glad I've stirred up all the questions, which I will try to get to.  I hope you don't feel I am any kind of authority on coding JavaScript or React.

Actually I hadn't got as far as discussing the OOP paradigm in relation to this.   

I can tell you that coming from a Java career into doing rich client side applications in JavaScript years ago, I began by trying to trying to impose classical inheritance on the code.  Now I know better, and understand all the problems with classical inheritance.

Again, if you know all this forgive me, but I understand you're newish to Javascript, so I don't know how much reading you've done into the language.

Before the days of React, you already had prototype inheritance and composition, and these still apply to react, but there is a strong composition model that comes with React.

I will try to get my head around how this specifically relates to this project ASAP, but in the mean time, I think [this page](https://medium.com/javascript-scene/master-the-javascript-interview-what-s-the-difference-between-class-prototypal-inheritance-e4cd0a7562e9) and [this page](https://medium.com/javascript-scene/common-misconceptions-about-inheritance-in-javascript-d5d9bab29b0a) are good primers.

jeffeb3, 2017-07-06:
There are some simple things that are still bothering me, like when do I use a semicolon, and what is the difference between var and let...

jeffeb3, 2017-07-06:
I read those pages. Great info. I really like the video about how composition > inheritance:
https://youtu.be/wfMtDGfHWpA

I also read about redux/flux/MobX:
https://medium.com/javascript-scene/10-tips-for-better-redux-architecture-69250425af44

I think we need something like this. I think the "full fledged" redux is not that much work, and will work well. MobX seems a little goofy to me, and I'm not sure how to merge that with the factory function composition the other articles are advocating. So I think we should use redux. WDYT? I know there are a few problems with things not updating currently if you change settings in a certain order, and I'd like to try making some localStorage so that settings (especially gcode ones) aren't destroyed with a refresh, so I think a state store is a good idea.

jamesd256, 2017-07-07:
We kicked off a big project at work a few months back, and spent a lot of time planning the architecture.

We were interested in binding react components to back end data at the outset, so different requirements here.  But we ended up with the following stack

React <- Using commons-chunk for bundling
Redux <- State management (as you know)
Apollo <- GraphQL client
GraphQL <- Obect persistence

It is a very elegant stack to work with and things have gone smoothly.  We haven't needed much re-use yet, but Sandify is provoking thoughts about that.

Redux is small and lightweight and very good at what it does.  Having a state management layer is a very good idea which will pay off very well once complexity grows.

jamesd256, 2017-07-07:
Initially we looked at Relay, which is basically the simple way to data bind react components.

The reason we didn't go with that, is that Relay is not really designed as a pure object data persistence layer.  Since we wanted to be able to define object graphs on the and seamlessly handle their persistence, we needed something  else for that, hence Apollo. 

And since Apollo can handle the duties also of dealing with react component data in the same way as relay, using relay was surplus altogether.

For visual elements which do not need back end persistence, relay and react is all we use.  But let's say you wanted to persist the state of the forms for a logged in user to a backend, it would be easy with this stack to add that in later.

jamesd256, 2017-07-07:
I have spent a little time reading about MobX vs Redux.  For a number of reasons, I feel Redux is more forward looking and simpler to work with.  Possibly the case that MobX will handle organising state store at scale, but that is just my feeling from looking at the way it does state storage.

Certainly Redux will do the job, and comes highly endorsed by the React team. 

1 more vote for Redux for me

https://www.robinwieruch.de/redux-mobx-confusion/

jeffeb3, 2017-07-07:
Redux it is then.

closed 2018-11-16 by jeffeb3