jeffeb3/sandify #28: Line coloring

jeffeb3, 2017-07-08:
Would it make sense to draw each line in a slightly different color? The progression of colors would show which lines get overwritten by later lines.  It also would make the displayed shape a bizarre number of colors.  I need to spend more time with github to figure out the mechanics of actually making the change in a branch.
 -- BT
https://www.vicious1.com/forum/topic/does-this-count-as-a-build/page/6/#post-37730

jeffeb3, 2017-07-08:
I was thinking I wanted to do this. @jamesd256 also is working on a way to visualize the sand better. I was thinking of using a color scheme like matlab's "jet". Matplotlib has a bunch of colormaps: http://matplotlib.org/examples/color/colormaps_reference.html

I could just count the number of vertices, and color each one based on it's percentage through the color line. Or I could use distance traveled (I want to accumulate distance traveled to, so I can provide some awful time estimates like everyone else). You could also bounce from one end of the color line to the other if you thought you weren't going to have enough definition between loops.

I also toyed with making the line thicker, but without any distinction, it just made it impossible to see the lines.

Any solution would have to be able to work on our array of vertices since the drawing code won't have any notion of loops from the original Transform.js component. Alternatively, we could make the vertices an array of arrays, and let the inputs handle what they would consider a single "layer"...

jamesd256, 2017-07-11:
I tried a few things with canvas, such as gradients and shadows.

Shadows are promising, but keen to explore all the stuff you mention too.  I'll post on the vicious1 forum to show what I've got.

jamesd256, 2017-07-11:
By doing begin path and stroke correctly around the background lines and the drawing of the vertices, I was able to get a much nicer shadow effect.  

However, this method is always going to be limited.  You really need to see the shadow effect of one line over another for each and every vertex, which won't work for performance reasons.

jeffeb3, 2020-04-05:
The current preview does a good job of showing which direction the path is taking. I think this is good.

closed 2020-04-05 by jeffeb3