jeffeb3/sandify #39: Dynamic or custom shapes

jeffeb3, 2017-07-20:
Here's a code snippet that I added to Transform.js to create a polygon of arbitrary number of sides.

```
    var poly_points = []
    var numsides = 8
    var angle = 360.0 / numsides
    var poly_scale = 1.0
    for (var i=0; i&lt;numsides; i++) {
          var radians = angle * Math.PI / 180.0
      poly_points.push(Vertex(poly_scale * (i + 1) * Math.cos(radians * i), poly_scale * (i + 1) * Math.sin(radians * i)))
    }
```

Jeff, it’s really a special case for when you are working with a regular polygon. The turtle stuff will be much more useful for non-regular shapes (conic sections, hearts and eggs, fractals). It could easily replace the code that’s in there for the triangle and square. But then it might make sense to just do them all with turtles…

How about we do two types of regular polygons, the one I’m playing with and another that alternates between ‘in’ and ‘out’ vertices, that would allow for multi-pointed stars. Leave turtle for the odd shapes.

From:
https://www.vicious1.com/forum/topic/does-this-count-as-a-build/page/8/#post-38887

jeffeb3, 2017-07-20:
I know that squares and triangles can be made with turtle, and that's perfectly fine. I have promised myself that I will get the turtle editor working before I do any more changes to the shapes. But, I also would like some sort of "gateway editing" where someone can design their own pattern without typing anything on the keyboard. Something just to get someone hooked. As a different mode, there is the turtle editor, which will allow much finer control of the patterns, and creativity in new ways. James and I came to that vision a while ago, of having turtle and an online editor being the path between clicking on shapes and writing javascript.

I would also like People Who Know What They Are Doing to contribute new, interesting input modes. You will be the guinea pig for that, it seems. Your dynamic shapes should either be merged in with the existing shapes (on that same tab) or brought it as a completely different tab. The spin and grow stuff is currently intricately linked with the current shapes.

billsey, 2017-07-20:
[image]
In regards to the 'Growing Thing', as you can see here there's a jump in the start of the left side for each rotation instead of a smaller movement on each side. My pull is from the origin/master. My code currently shows the same results, except I am rotating center to east instead of center to northwest.

    var poly_points = []
	var num_sides = 4
    var poly_scale = 1.0
    for (var i=0; i<num_sides; i++) {
      angle = Math.PI * 2.0 / num_sides * i
      poly_points.push(Vertex(poly_scale * Math.cos(angle), poly_scale * Math.sin(angle)))
    }

jeffeb3, 2017-07-20:
That's definitely not master. Master is yellow now. https://jeffeb3.github.io/sandify/

You probably need to merge in the more recent changes of master into your branch.

jeffeb3, 2017-07-20:
w.r.t. dynamic shapes. If you want, I can make a dynamic shape, based on your N sided polygon, and then you can see how I did that and then create the N sided start. WDYT?

billsey, 2017-07-20:
Weird, when I sync or pull it shows nothing newer than 7/4.

Ah, figured it out... I'd somehow created a copy in my github page that I was syncing to, instead of syncing to the real thing. Now it complains about missing the react stuff. No 'redux' or 'react-redux'.

billsey, 2017-07-20:
Just had to install those two packages... It is looking correct now, here's the 13 sided example:
[image]

jeffeb3, 2017-07-20:
Very cool. Can you fork and push that code?

jeffeb3, 2017-07-20:
Also, anytime the package.json adds a new dep, you'll have to run `npm install` from the toplevel. There will be a 'victor' package coming very soon.

billsey, 2017-07-20:
Think I've got it... Created a branch with the code and committed it. Let me know what I've screwed up. :)

jeffeb3, 2017-07-20:
A commmit is just a local thing. You have to push it too. You probably can't push to this repo though, you'll have to fork it.

billsey, 2017-07-20:
Yep, I can't push... I fork from the github interface like I did accidentally before (clone repository)?

jeffeb3, 2017-07-20:
Yeah, you'll fork (through the web interface), then change your origin (on your machine), then push (also on your machine).

billsey, 2017-07-20:
OK, I think I've got it... https://github.com/billsey/sandify.git should have the latest. I'm still not sure how to pull the sidesValue variable into ShapeList, perhaps you can enlighten me?

jeffeb3, 2017-07-20:
Got it. PR #40

closed 2018-11-16 by jeffeb3