jeffeb3/sandify #45: Versions/News/History

jeffeb3, 2017-08-12:
Now that ZXY is in the wild, people will start to really "need" this tool. I don't think it's feasible to have previous versions working, but I wouldn't mind having a developer/staging branch and a stable/deployed branch. Then the next thing is to add version numbers, and figure out a way to post a changelog and a way to announce any new features to users.

Anyone have any thoughts on a good way to do that?

billsey, 2017-08-15:
Doesn't GitHub do that for you? Or are you thinking of pushing updates to users who don't have GitHub accounts and aren't watching the project? Version numbers should be easy, and IIRC in the past I've seen them auto-generated when master is updated, though the version numbering is best handled then by x.y.z.buildno instead of x.y.z. You just keep your development branches active and only update master when you merge in changes from a development branch. A change log is a function of Git itself, though it might be better to have a text document as part of master that gets manually updated instead of putting all the little changes in automatically.

How do we visualize this being actually used? Will each user have a PC running Sandify that outputs gcode they then put into the ZenXY card slot? Or will there be a public website they can use to generate the code? It seems like the latter would be easier to deal with. The public site (https://vicious1.com/Zenify?) would be automagically updated when the project is updated. Users who want to play with it locally can, but they'd be encouraged to handle updates and such themselves manually when someone official posts that there's a new version in the appropriate forum. That's how we handled game servers in Mangband...

jeffeb3, 2017-08-15:
I guess what I was thinking is that the webpage itself would be version, and have a user relatable changelog. But, even after reviews, I don't want to break the published version of sandify, so ideally there would be some users willing to run it on their home computers to test a developer branch and the GitHub.io site would be more stable. I was thinking users might want to subscribe to updates and I could email out when it gets changed. So if we added the jpeg to traveling salesman thing, then when it went from stable to master, it would let everyone know, and they could use it

James and I were thinking it would start as a shared website that generates gcode for you to save. But ultimately, people would be able to install it similar to octoprint on a machine connected to the machine, and it would send the gcode live. That part has had 0 work done, but it was considered when we developed the arch.

jeffeb3, 2017-08-15:
Also, although sandify was created for ZenXY, I think it makes sense to keep an arms length away. Ryan doesn't have the resources to maintain the code himself and there are thousands of machines this should work for. Hopefully, some day, someone from those other camps will start contributing, and that should lead to better code for everyone.

closed 2018-11-21 by jeffeb3