jeffeb3/sandify #47: Orgin moves- Not an issue a request.

Allted, 2017-08-25:
Is it possible to have it continue from the end point along the edges to the lower left corner? Or maybe out to the nearest edge, then along them home? If it is rough I do not mind adding the 2-3 lines in by hand, just an idea.

If run in reverse this lets us home the machine and not over run the current pattern, without having to add in 2-3 lines of code by hand.

If run from the center it would make it end at the home point.

This just let you stack designs more easily. As it is I run the first in reverse (from "home" to the center) then the next from the center out by cutting and pasting them together. This seems to keep my table media really level.

jeffeb3, 2017-08-25:
This would be a decent option. I agree that it would be useful. I wouldn't want it to be always on. The pattern generation is open ended. It doesn't know about the limits and it doesn't stop when it reaches them. So this would have to compute the right path based on the last vertex and the machine limits. I think that's pretty simple, just compute which side is closest, move there and then trace around to the start.

jeffeb3, 2018-11-22:
The more I play with this, the more this is becoming my biggest problem.

billsey, 2018-11-22:
What are you running into? I'd guess dealing with patterns that are smaller than the available space, so you end somewhere other than an edge or the center? Can you do a comparison between the center position and current position to determine the closest edge, then move there before tracking around the edges?

jeffeb3, 2018-11-24:
I think I'm just wanting the patterns to all start and stop at the same place. So if the star pattern expands all the way to the edge, but ends up along the right, it would move out to the right edge, and then along the edge to the lower left. 

Then I could turn off the machine, and home later without any trouble. 

I'll make it an option, similar to the polar force endpoints option. Should I force lower left? I would want it to be flexible, but I can't really imagine why someone would want a different corner. I'll try a few things.

jeffeb3, 2018-11-24:
As for detemining which side to drive toward, I'll compare the last coordinate to the lines y=mx and y=-mx. M would be the ratio of y size to x size. That should tell me which side is closest.

Allted, 2018-11-24:
I was actually thinking about this yesterday. Since we home Y before X all you need to do is make sure you do not stop at the top edge. Any of the three others are much more ideal. Right now I just edit the code by hand, looking at the last position and adding a line if needed.
For a center ending pattern it would be nice to for the pattern to stop in the exact center of the table so you can power down and know a center starting pattern is in the exact center at the start. Currently I only run a center starting pattern without powering down to avoid hitting a limit.

Allted, 2018-11-24:
Ah....right the Roundies.

jeffeb3, 2018-11-24:
Well, the polar machine has a decent function for that, but there have been some other xy machines using sandify. Like the one you found on HAD.

Allted, 2018-11-24:
That is why you are the smart one! I have a narrow field of view.

jeffeb3, 2018-12-05:
I have this working in a branch. It wasn't too tough, I just kepy making stupid mistakes because I was watching TV at the same time :). I ended up scaling the last point so it was on the circle made by the 4 corners. Then it's easy to compare to min_x, max_x, etc. to find which side it went out on, then I always trace counter clockwise until I reach the origin. I added buttons to choose the origin, and other than them being a little ugly, they are working.