jeffeb3/sandify #50: Looping gcode

billsey, 2017-09-19:
Assuming we have direct downloads of gcode  (#49) it would potentially be helpful to allow the user to specify that the patterns be repeated after completion, or even that several patterns could be chained and the chain repeated. I know this one is likely to be complex, since we'd have to deal with moving from one pattern to another, preferably without walking all over the previous pattern in the process, but it'd sure make it easier to sell a ZenXY system into a business if they didn't have to manually trigger each 'print'.

jeffeb3, 2017-09-19:
I don't know about looping. Maybe that would work, but I think having an area where different patterns are assembled into a program makes a lot of sense. I wish I was better at creating the layouts for this kind of thing, because I think the actual work to take a pattern and add it to an assembly area isn't that hard, it's just finding a way to illustrate what's happening to a user is the difficult part.

billsey, 2018-11-16:
Just to capture here:
```
G28 X Y

M32 P !StarO.gco#
M32 P !StarI.gco#
M32 P !SquareO.gco#
M32 P !SquareI.gco#
M32 P !LogoO.gco#
M32 P !LogoI.gco#
M32 P !CircleO.gco#
M32 P !CircleI.gco#
M32 P !TriangO.gco#
M32 P !TriangI.gco#
M32 P !WipeOO.gco#
```

mrehorst, 2023-11-26:
Maybe you need a different controller board...

I use a Duet 2 WiFi 3D printer controller board in my sand table. The board runs RepRap firmware that has the ability to run macros, including running one at power up. When I am going to take my table to a MakerFaire I create a macro file that contains pointers to a bunch of patterns to run in sequence (about 10 hours worth of patterns) and call that macro from the config file that runs each time the machine powers up. That frees me from having to take a computer or trying to mess with wifi, etc., at the MakerFaire and makes sure that if there's any problem, all I have to do is cycle power and the table will run.

bobnik, 2024-10-16:
@jeffeb3 though it's not via "looping gcode", is this issue resolved _enough_ by our multi-layer feature so that it can be closed?