jeffeb3/sandify #53: Support for circle tables

anthonywebb, 2017-12-22:
Hi there, I am contemplating having a square table, with only a circle area exposed in the middle.  Right now in the software the limits are defined as a rectangle.  Would it be impossible to defined the usable space as a circle?  I'm not opposed to rolling up my sleeves and sending you a pull request, would just like to know ahead of time if you see this as even a possibility?

jeffeb3, 2017-12-22:
It's no big deal to add it to the ui. The limit code is already a bit fragile though. It would be easier to handle a circle. 

I think I would just write an entire new function for filtering on a circle. Changing points outside to the edge would be easy, but then figuring out where to trace would take a little work. Just like with the rectangle, you'd have to handle the case of an exit or reentrance separately from cases where the lines stayed inside or stayed outside.

It's not too bad, and it's a reasonable feature to add. It is not easy.

jeffeb3, 2018-11-18:
Done. Finally.

closed 2018-11-18 by jeffeb3