jeffeb3/sandify #56: Sandify local

dzech, 2018-07-14:
Hello,
I wanted to run SANDIFY local on my WINDOWS 7 computer but it don't work.

I installed NPM and NODE.JS
Download and unpacked the ZIP file
Run "npm install package"
Run "npm start"
[2018-07-14T16_22_02_462Z-debug.log](https://github.com/jeffeb3/sandify/files/2195114/2018-07-14T16_22_02_462Z-debug.log)

How can I fix it?

jeffeb3, 2018-07-14:
Hmmm. I have no idea. I haven't used windows in a long time and none of that output seemed familiar to me. Sorry.

closed 2018-11-15 by jeffeb3

Mahelumi, 2018-11-16:
Hey .. The JS just work fine . Where did you install the npm. You should be
install that in the source folder of the sandify. Ex. 1) open cmd with
admin rights
2) navigate to the extracted source folder like D:/sandify folder/
3) hit npm install . Also check there are many dependent files installed in
that folder other that sandify source. If not the JS wont run
4 ) now npm start.  Weeeee it will take you to your browser..