jeffeb3/sandify #58: new UI generating coordinates beyond set limits

mrehorst, 2018-11-17:
I love the updated UI- it's much easier to generate patterns quickly by scrolling the parameter values.  I generated a bunch of gcode today and for some reason the files contain coordinates far beyond the limits I set on the UI.

I set X limits to min 5, max 730, and Y to min 5, max 1560.

Here's a start of one of the files (I added file name and speed):

; r_l_p7.gcode
G01 X3227.500 Y26459.896 F30000
G01 X3227.500 Y26459.896
G01 X3227.500 Y26557.500
G01 X3227.500 Y26557.500
G01 X2502.500 Y26557.500
G01 X2502.500 Y26557.500
G01 X2502.500 Y25727.604
G01 X2502.500 Y25727.604
G01 X2502.500 Y25002.500
G01 X2502.500 Y25002.500
G01 X3227.500 Y25002.500
G01 X3227.500 Y25002.500
G01 X3227.500

mrehorst, 2018-11-17:
here's an sample of the inputs:

[image]

and here's the start of the gcode file it generated:

G01 X3227.500 Y25780.000
G01 X3227.500 Y25780.000
G01 X3227.500 Y25868.859
G01 X3227.500 Y25868.859
G01 X3227.500 Y25957.433
G01 X3227.500 Y25957.433
G01 X3227.500 Y26045.506
G01 X3227.500 Y26045.506
G01 X3227.500 Y26132.864
G01 X3227.500 Y26132.864
G01 X3227.500 Y26219.295
G01 X3227.500 Y26219.295
G01 X3227.500 Y26304.590
G01 X3227.500 Y26304.590
G01 X3227.500 Y26388.542
G01 X3227.500 Y26388.542
G01 X3227.500 Y26470.947
G01 X3227.500 Y26470.947
G01 X3227.500 Y26551.607
G01 X3227.500 Y26551.607
G01 X3227.500 Y26557.500
G01 X3227.500 Y26557.500
G01 X3227.500 Y26557.500
G01 X3227.500 Y26557.500
G01 X3227.500 Y26557.500
G01 X3227.500 Y26557.500

jeffeb3, 2018-11-18:
@mrehorst, Thanks for reporting this. I tried reproducing, this and I don't see the same result:
[sandify.txt](https://github.com/jeffeb3/sandify/files/2592496/sandify.txt)

It always downloads it as "sandify.gcode". Maybe your browser downloaded it as `sandify(1).gcode` instead, and this is an older one?

Can you post the gcode (github doesn't like the extension, but renaming it to .txt works)?

mrehorst, 2018-11-18:
Yes, it downloads as sandify.gcode, and then I change the file name and add
the new name inside the file.  I stack patterns in a single, large file
that will run continuously for 12 hours or more, usually in sequences of
two or three patterns followed by a wipe.  By adding the file names I can
locate them in the big file in case I need to make edits or change patterns
(some patterns don't look very nice in sequence).  I save the files with
the start and end locations and a pattern number (so l_r_p4 means pattern 4
starts on the left and ends on the right).  I also save the image of the
pattern (example: p4.png) for reference when planning sequences.  I try to
sequence the files so that each file starts where the previous one ended so
the ball doesn't charge across the table and leave a line in the middle of
the last pattern.

While I have your eyes, if I may make a suggestion...  it would be really
nice if the parameters used to generate a pattern were automatically stored
in comments in the gcode file.  That would make it possible to recreate the
file and tweak it a little if needed.  It would also make it easy to post
interesting patterns for others to use.

I would volunteer to do the programming, but I haven't programmed anything
for about 20 years - I first learned Fortran IV on a PDP11(?) back in the
early 70s, and did some PL1 programming using punched cards when I was in
college in the late 70s.

I'll post the full gcode file with a .txt extension when I get to my other
computer.

Thanks again!

Mark Rehorst, DMD
262-321-8264

https://drmrehorst.blogspot.com/
http://mark.rehorst.com

jeffeb3, 2018-11-18:
Storing the details in the header is a good idea. Similar to how slic3r does it. I will add a new issue with that. I think I know a good place to do that so I'm not making a ton of code to copy the setting to the gcode parts from the shape parts. I think the ';' is more common (for Marlin/grbl machines).

I just added some stuff for polar machines, so I probably broke something else. I spent a good amount of time trying to reproduce this error, and I couldn't. Please don't be discouraged. You're providing valuable testing for me.

I also just added a filename input to the save dialog. That should make your naming thing go easier.

mrehorst, 2018-11-18:
OK, I tried again and it did the same thing.  Then I closed the browser tab, opened a new one, and pointed it at Sandify, and noticed my X and Y min and max were retained (cookie?).  I tried again and it generated coordinates beyond the set boundaries.  Then I closed the browser (chrome), reopened it, went to sandify, and my X and Y min max were gone.  I reentered them and generated a pattern and it worked fine.  It seems like there may be some issue with the way I've configured my browser, or with the way the program interacts with it.  Or maybe you were updating things while I was playing with it and that created the problem so it really isn't a problem after all...

Here are the settings:
[image]

and here's the file, which looks identical to the other failed file:

[r_c_p01.txt](https://github.com/jeffeb3/sandify/files/2592550/r_c_p01.txt)

jeffeb3, 2018-11-21:
Those are very strange coordinates. IIRC, the limits code works in the coordinates where the center is 0,0 (which is what sandify uses internally, until it's time to write out the gcode). I wonder if some of that centering code is broken with some inputs, like max being less than min or something. It's weird that they are all that large. Have you seen this problem again?

mrehorst, 2018-11-21:
I haven't run it for a day or two...      work...

I'll get back to it today and see if I can make it do weird things again.

https://drmrehorst.blogspot.com/
http://mark.rehorst.com

jeffeb3, 2018-11-21:
I don't know if it's using cookies or what. It's based on the localStorage library from JavaScript, and it's probably cookies.

I found the bug though. The code to read that from the storage wasn't converting it from a string to a number, so when I did math with it `center = (max - min) / 2.0`, it was politely doing what I asked without mentioning that subtracting two strings makes no sense. I have deployed the fix (I hope).

closed 2018-11-21 by jeffeb3

mrehorst, 2018-11-22:
It appears to be working very well!  I haven't been able to break it
again.  The new changes make it a lot easier to stack patterns in sequence,
but I have a few additional suggestions... (of course!)

1) put the pattern file name at the end of the file as well as at the
start- this would be useful for the way I manually stack the pattern files
(I don't know how useful it would be for others), all in one long gcode
file.  The only reason I do it that way is because my smoothieboard
controller firmware has a bug and can't call gcode files from within a
gcode file (though
https://www.reprap.org/wiki/G-code#M32:_Select_file_and_start_SD_print says
it can, it can't).  Some 3d printer controller firmware like the RepRap
firmware and Marlin can, which means the sequence file would simply consist
of a series of calls (M32) to pattern files in the same directory on the SD
card.  I would guess that most tables are driven by 3D printer controllers
running Marlin since those controllers are the cheapest ones out there.

When I name the files, I include the start and end locations of the pattern
with a serial number for the pattern.  So a pattern that starts on the left
side and ends at the top right corner will be named something like
l_tr_p04.gcode.  When I am assembling sequences of patterns, I try to start
the next pattern where the last one ended.  So using the example named, I
would select the next pattern in the sequence to start at either the top or
right side of the table.  That's why having the filename at the start and
end of the file is useful.  But maybe I'm the only one who sequences files
this way.

For each pattern I generate, I usually save 3 files.  One is the image of
the pattern so that when I'm sequencing them I step through the images and
look for patterns I think might look good in sequence and have the right
start and end locations. I also store forward and reverse gcode files. In
the example above, I'd store a p04.png file along with the l_tr_p04.gcode
and tr_l_p04.gcode files.  It would probably be better if I also included
the start and end locations in the .png file names, too.

It would be nice if the pattern generator would let me enter a pattern
number like 04, or p04, circle05, etc., and then the generator would attach
the start and end designators and save the gcode file with all those
designators in the file name, and save both the forward and backward
pattern files.  So I'd generate one pattern that I like, then hit a button
to save it, and it would save the image file and the two gcode files all
with the names assembled from the start, end, base shape, and a unique
serial number.  That would make creating a groups of files to sequence and
the sequencing of them easier to do manually, and would even make it easier
to write a program to generate sequences of patterns automatically.

2) is there some way that the settings could be combined into a sort of
serial number/hash that could be stored in the file and then manually
entered (copy/paste) into the web page to reproduce a pattern?  With
something like that, instead of posting tables/lists of settings, users
could post the serial/hash number so others could reproduce the pattern.  I
think the human readable list of settings should also remain in the pattern
files.  A hash formula might have to allow for a bunch of extra, currently
nonexistent variables for the addition of other pattern types in the future.

3) is there any way to control the "granularity" of the curves?  I have
found that in some of the patterns based on a circle, the individual
segments that make up the curve start to become visible in the pattern in
the sand, making it look more like a polygon than a curve. That seems to
happen for large radius curves more than small, so maybe if you changed it,
the small curves would turn into too many G1 commands...Maybe granularity
could be set as a function of radius of curvature.

4) looking at the gcode, I often see G1 commands repeated multiple times.
I assume this is due to rounding the coordinate values.  Can the duplicate
G1 commands be scrubbed from the gcode file or maybe tested before they get
added to the file (if the current G1 command is the same as the last one,
don't add it to the file)?  I don't know how the 3D printer controller
firmware handles multiple identical commands- do they all get stacked in
the buffer or does the firmware ignore multiples of the same command?  This
would be a low priority because giant gcode files generally aren't a
problem for the controller firmware.

5) finally, there doesn't seem to be any gcode optimization.  The ball
often travels great distances along the edge of the table to get to the
next line on the table surface which is right next to the last line.  I'm
sure it would be a major undertaking to optimize the gcode to minimize the
path traveled by the ball for any given pattern.  That would also change
the way the pattern appears and maybe not always for the better.  For
example, some patterns start at the center and completely overwrite the
stuff at the center by the time they are finished.  It would look different
if you "optimized" the pattern to minimize ball travel- that stuff at the
center that was overwritten in "nonoptimized" gcode, might end up being the
last thing drawn in "optimized" gcode.  Still, optimized gcode would draw
faster, and entail less wear and tear on the table mechanism...

Sorry for writing a novel...  maybe I should just post these things to
github..

Thanks for all the great and very fast work you've done!

Mark Rehorst, DMD
262-321-8264

https://drmrehorst.blogspot.com/
http://mark.rehorst.com

jeffeb3, 2018-11-22:
> 1) put the pattern file name at the end of the file as well as at the start- this would be useful for the way I manually stack the pattern files (I don't know how useful it would be for others), all [...]

Regarding M32, they are looking for dos names, which is something like only the first 5 characters. Try renaming them to something small, like test.gco and running M32 again. There is a way to query the files on a card too, and I think that returns something that will let you translate between the dos names and the full names.

Regarding the rest, I was thinking that I might end up doing some simple pattern matching, so you could put {start_quadrant}_{shape}_{end_quadrant} and it would replace that with 0_polygon_3 or something. Anything more specific feels like it would be more confusing than helpful. I'd like for someone to be able to use this, and not be confused without reading any wiki or manual about it.

> 2) is there some way that the settings could be combined into a sort of serial number/hash that could be stored in the file and then manually entered (copy/paste) into the web page to reproduce a [...]

The real trouble with that is that the definition of the hash will have to change over time, as I add in more shapes or inputs, the hash will get more and more complicated. I haven't decided if I'll add something that can just read in the human readable format. It's got the same problem, but it's probably about the same amount of work. 

> 3) is there any way to control the "granularity" of the curves? I have found that in some of the patterns based on a circle, the individual segments that make up the curve start to become visible in [...]

The difficulty with that is, that I'm not really supporting curves. I create the original circle from 128 points, and then those lines are scaled and rotated and looped. By the time the circle is huge, I no longer have the information that it used to be a circle. I'm grappling with a similar issue on the polar tables, because some of those are 10 feet wide! I could add an option for changing the number of vertices in a circle, or I could make it 1024, and then downsample the points when the circle gets very close in. I could also have a concept of an arc in the vertices, but that's going to really screw up the code. It would be hours of work for very little effective difference (but that might make sense as the problems get harder to solve without arcs).

> 4) looking at the gcode, I often see G1 commands repeated multiple times. I assume this is due to rounding the coordinate values. Can the duplicate G1 commands be scrubbed from the gcode file or [...]

Good point. Marlin only accepts 3 decimal places, and that's probably waaay too fine. I should probably round down to some reasonable value, and then compare the adjacent vertices and delete the duplicate, as you say.

> 5) finally, there doesn't seem to be any gcode optimization. The ball often travels great distances along the edge of the table to get to the next line on the table surface which is right next to [...]

I have no idea how to solve a problem like this. I'd love some help if someone wants to take this on. I personally really like the idea that the sand table writes over itself. It gives it a more 3D appearance when it's running. As for driving around the outside too much, it's probably part of the limit enforcement code. The code isn't an artist, it's a set of rules, and they sometimes drive too much in the wrong way. I'd love some help with that too. It's not the most fun part to work on (making new inputs is).

As a side note on github issues:
a) I think you're replying via email, and your signature includes your phone number. These are getting posted on the internet, in public, so you probably don't want that.
b) It would be easier for me to inventory these if they are separate issues. I've been sitting down to work on sandify, and going straight to the issues page, and looking for things I think I can get done. If these were each separate issues, I could label them, or organize them in a way that would help me find the details faster.
c) This specific issue is closed, because it's fixed, so that's another reason to make new issues instead of using this one. This one won't even show up in the issues tracker anymore, because it's fixed.

Thank you for the novel. I appreciate the experience you're sharing. Even the stuff you've suggested that I think is too hard for me to tackle right now is giving me ideas on how to improve things.