jeffeb3/sandify #62: Import gcode files

jeffeb3, 2018-11-22:
Similar to the way the thetarho import lets gcode users convert thr files, thr users should be able to convert gcode files.

This is a little more complicated, because I'll need to parse the gcode for the features, which will mean at least supporting everything G0/1/2/3 does. But I'd like to do that anyway. Maybe someone knows of a good gcode javascript library?

In the UI, it probably makes sense to have it in the Code tab, and have the import button handle both, with some auto detection based on the extension, or the content. The rest shouldn't change.

billsey, 2018-11-22:
No idea how good they are, but I see several on https://libraries.io/search?keywords=gcode&languages=JavaScript

jeffeb3, 2018-11-24:
Thanks Bill! This one looks pretty good (MIT, and it's part of cnc.js): https://github.com/cncjs/gcode-toolpath/blob/master/README.md

closed 2020-04-23 by jeffeb3