jeffeb3/sandify #64: Drawing Text

billsey, 2018-12-02:
I don't know, maybe this should be in #57. I'd like to be able to enter some text into a text box and have it rendered on the table. This would be pretty complex to do, I believe, even if you limit to a single font style. But it's really open the package to commercial uses. Perhaps with the added functionality of adding line feeds... "Buy Beer/nHere" would display as:
```
Buy Beer
  Here
```

jeffeb3, 2018-12-02:
So, in case you couldn't tell, I'm wording this thinking thataybe you'd want to do that Bill :). If you can make a couple letters to start, I can build the rest around that and we can either split up the alphabet or something to get to the finish line. I'm assuming there isn't a good way to cheat.

One way I was thinking we could cheat is to use EstlCAM to cam up one letter at a time and export it as gcode, then convert it to vertices.

jeffeb3, 2018-12-02:
I was messing with this the other day:
[jeffeb3.txt](https://github.com/jeffeb3/sandify/files/2637525/jeffeb3.txt)

billsey, 2018-12-02:
What would you think of starting with a sixteen segment display font type? It should be easier to work with initially than a normal stroke font with arcs and lines intermixed. I also suggest the underline be under the letter itself so it's tougher to get confused with E vs F. Though EstlCAM creating a set of lookup tables works as well... We could fix on a generic font first and only go full Gothic after there's a better way to draw them. :) Hmm, maybe look at how Postscript does fonts? Aren't there open source Postscript implementations?

jeffeb3, 2018-12-03:
I think any set of vertices will work. Starting with a 16 segment would be fine. I was thinking there could even be thicker letters with a bit of stroke magic. Any set will test everything else, then it's just a matter of doing the art of writing a font.

I think taking the shape information from any normal font type would probably be as much work as writing CAM software. Maybe harder because it has to be 100% automatic. :). There needs to be something thatakes the whole thing actually traceable, so I think some manual fidgiting is necessary.

billsey, 2018-12-03:
http://www.imajeenyus.com/computer/20150110_single_line_fonts/index.shtml has some pointers to Hershey fonts implemented as python extensions to Inkscape. https://github.com/ixd-hof/HersheyFont has examples of using drawbots to place the font characters, including a bunch of font data files. More on Hershey fonts here: https://github.com/frankrolf/hershey-fonts and here: https://lwn.net/Articles/654819/. Here's another set of example code: http://www.ks.uiuc.edu/Research/vmd/doxygen/Hershey_8C-source.html, although it looks like he may have embedded the font in the code instead of loading the font data. The potential problem I see at first glance is that the Hershey fonts assume you can do a pen up command, and we can't do that without some really, really tricky magnetics. :)

billsey, 2018-12-03:
Oh, and I'm finding lots of examples for Turtle too. :)

jeffeb3, 2018-12-03:
Oh boy. That Hershey code is a delight! Brings me back. If you look through the text table, anything that has " R" (space + R) is a pen lift. That's quite a few. I'm guessing some of them are things like a 't' picking up instead of tracing inside the letter to make the complex shape.

jeffeb3, 2018-12-03:
The Hershey script would be really cool. I don't really like the license with them, and it will take some manual fidgiting, but that's a great start.

jeffeb3, 2018-12-03:
The asteroids code has a cool history too, and looks very easy to consume. It would be a good first try too:

https://github.com/osresearch/vst/blob/master/teensyv/asteroids_font.c

jeffeb3, 2018-12-03:
I'm assuming you want it to loop and spin like other shapes, right? You don't just want it to draw it once, right?

Mahelumi, 2018-12-03:
Looking to be the great thread. Jeff you have done a great job with the new
update and worked with many designs including various text with different
font styles. To all kindly go through all usefull links in the sandify
website. Dont get the things complicated. Ill share all my doodles soon

Thanks

jeffeb3, 2018-12-03:
Darnit! That asteroids font is GPL2, so I'd have to make this whole project GPL2 to bring it in.

jeffeb3, 2018-12-03:
I pushed a branch with some skeleton code. I based the letters I did off of a 16 segment, like you suggested. I didn't do that many, just enough to try some stuff out. It's pretty hideous, and I can already tell that munging the fonts is going to be more art than math. I still don't really like how I forced them to connect, but I'm not sure what else I can do, really. I would like to have a way to support more than one font, and I would love it if it were easy for others to create fonts, but I think that's probably a big stretch.

There are two other problems I found when playing with this:
1) When the letters grow or spin, each vertex grows or spins a bit more than the previous one. This means some of the back tracking isn't lining up. I think I need to change the vertices from an array, to an array of arrays. In the shapes, each sub array will have one vertex. In the letters, each letter will be it's own subarray, and I'll decide how much to grow/spin for each subarray, not each vertex.
2) When you hit 'enter', the whole page updates. At least it does in my browser. I think this is because it's technically a form, and forms gotta go somewhere. IDK. It's annoying.

I didn't try to fix the \n problem. I'll try that next.

jeffeb3, 2018-12-03:
The branch is called 'text'
https://github.com/jeffeb3/sandify/tree/text

billsey, 2018-12-04:
Actually, like every other design, I'd likely use it as a single draw of the text sometimes and as a rotating, expanding (or contracting) draw of the text other times. What I probably wouldn't need is rotating, expanding text that changes each run through. :)

billsey, 2018-12-09:
OK, I was putting some thought into this... I think we should start with just one font and all upper case to simplify the needs. Upper case letters do not have little dots above them like lower case i, so no need to pick up the pen anywhere. The letters are made from just a few primitives (straight line from point to point, curved line forming 1/4 circle or oval from point to point with the curve bulging into a direction (NE, NW, SE, SW), entry, exit point into the character). I believe a font such as the base font in the Hershey fonts can be drawn mapped to an 9x9 grid, where the ninth row is the 'underline' that all objects are drawn from and the ninth column provides the spacing to the next character. If the grid numbering is from 0,-1 at bottom left then:
'A' is: Enter character at 0,0, draw straight line to 3,7, draw straight line to 7,0, draw straight line to 5,3, draw straight line to 3,3, draw straight line to 0,0, draw straight line to base, draw underline to next character.
'B' is: Enter character at 0,0, draw straight line to 0,7, draw straight line to 3,7, draw curved line to 7,5 bulge to NE, draw curved line to 3,3 bulge to SE, draw straight line to 0,3, draw straight line to 3,3, draw curved line to 7,1 bulge to NE, draw curved line to 3,0 bulge to SE, draw straight line to 0,0, draw straight line to base, draw underline to next character.
The most difficult character I see is a Q which really wants to connect that tail into the middle of a curve, but could be drawn with the tail connecting to bottom center instead. Bulge directions could be represented by a 0:4 list starting at none, then NW and going clockwise. Entry point on all characters start at the bottom, again 0:4, so we could use an array of arrays for each character.
'A': { {0,0,0}, {3,7,0}, {7,0,0}, {5,3,0}, {3,3,0}, {0,0,0}, {0,-1,0}, {0,-1,8} }
'B': { {0,0,0}, {0,7,0}, {3,7,0}, (7,5,1}, {3,3,2}, {0,3,0}, {3,3,0}, {7,1,1}, {3,0,2}, {0,0,0}, {0,-1,0}, {0,-1,8} }
We can adjust the look by adjusting the end point of each draw by an asymmetrical factor. Square letters use 1:1, tall letters might be 0.5:1, wide letters might be 1:0.5. Eventually we do kerning by moving the start point for each letter based on the previous letter. A followed by M don't adjust, A followed by V adjusts closer, allowing overlap AVAV. I pretty much always kerns in both directions AI, III, IV.

jeffeb3, 2018-12-19:
OK, I've spent some time looking at your write up, and I like the idea of supporting just the quarter circles, one quadrant only curves. I implemented this, using your A and B, and there are a few things I'd change, but it's looking OK.

[image]

First off, the units of 0-8 are pretty brutal. I added a bunch of /8.0 or /4.0 everywhere to get them to scale right, and there's no reason not to use floating point. For example, I needed to test the other quadrants, so I drew an 'O' with this:

  'O': [ [3.5, 0, 0], [0, 3.5, 3], [3.5, 7, 4], [7.0, 3.5, 1], [3.5, 0, 2], [3.5, -1, 0], [8, -1, 0] ],

And I wanted it centered, so I used 3.5s. I like to think of it in terms of 2.0 high and 1.0 wide, but I'd also be OK with 1.0 and 1.0. I will still provide some mechanism to scale it eventually, but I think it will just be a lot easier to even write it up for each letter with a floating point scheme. The A didn't quite line up, but a floating point (even scaled to 8) would fix that.

Secondly, the NW, SW, SE, NE scheme isn't too bad, but if I already have the starting (previous) and end points, then all I really need is, is the circle in clockwise, or counter clockwise? I had to predict that based on the quadrant and one of the numbers (so NW, CW has a Y going down, NW CCW has a Y going up). I already wrote the code to properly decipher it, and if the quadrant is the easiest way, then that's OK, I can deal with that.

Can you use '[]'s instead of '{}'? It saves me one step.

Lost countly, At some point, pretty quick, I just convert all of this into vertices, so there's no real reason that we couldn't do this, and later support any other floating point font (except that you can't lift the pen, etc). I like how compact this format is though. 

The code to display your font is here: https://github.com/jeffeb3/sandify/blob/text/src/inputs/Fonts.js

billsey, 2018-12-19:
Well, according to [LifeWire](https://www.lifewire.com/aspect-ratio-table-common-fonts-3467385) the normal aspect ration is around 0.5, so your 2:1 is about right. :)

I started with 0-7 to make a 8x8 square, then remembered I had to duck under for the base so I didn't need to pick up a pen and I had to put a space between characters. We could use 0-6 for the character and 7 for the base line and the spacing if that would make it easier... I think it worthwhile that you can use letters that aren't completely symmetrical, as shown in the Bs above.

As for quadrant vs clock, I have no issue with using clockwise, counter clockwise and straight. Perhaps -1, 0, 1 with zero being straight, clockwise 1 and counter -1? Can't quite do bits, but chars aren't too nasty...

I'll get to work drawing letters on some graph paper and turning that into something more usable this wekend. Yes, I can use [] instead of {}, I guess I was 'C'ing too much.

jeffeb3, 2018-12-20:
The coordinates are going to be floating point, so there's no reason you have to stick to decimals. There are only 26 numbers. The whole alphabet will be much less than a single image, so no need to worry about space here. If you want slightly more than the middle, you can do 0.55.

billsey, 2018-12-27:
OK, so I tried these first couple of letters:
`  'C': [ [3.5,-1,0], [3.5,0,0], [0,3.5,4], [3.5,7,1], [7,6,2], [3.5,7,2], [0,3.5,1], [3.5,0,4], [7,3,3], [3.5,0,3], [3.5,-1,0], [8,-1,0] ],
  'D': [ [0,0,0], [0,7,0], [7,3.5,2], [0,0,3], [0,-1,0], [8,-1,0] ],
`
And they draw all weird on me. :) Can you see anything I'm doing wrong, other than not getting enough sleep this week? I'm also open to not having the C curve quite so much in toward the middle on the right side if that looks better.

billsey, 2018-12-31:
[image]

billsey, 2018-12-31:
Dang I was going great on the numbers (slightly narrower than  letters) until I got to '3' when things broke. 3, 4 and 5 do nothing...

jeffeb3, 2018-12-31:
Did you share the letters somewhere?

billsey, 2019-01-01:
I checked them in I thought...

Bill

Sent from my Verizon, Samsung Galaxy smartphone

jeffeb3, 2019-01-02:
I've tested your branch, and I have a few small fixes pushed, but I basically think this is ready to go. I'm trying this on my actual machine, and I have a couple other cached changes to include.

jeffeb3, 2019-01-02:
I'll be honest, it's not great, but I think it is an excellent start and hopefully, it will be a good foothold for future text based stuff.

jeffeb3, 2019-01-02:
I'm going to close this issue, but make a few more for smaller tasks, and label them with a text label.

Some things you and I have noticed:

1) We need \n, <cr> or whatever.
2) We need more letters and possibly more fonts. International currencies for one thing.
3) We need to treat each letter as it's own object when it comes to scaling or rotating it. Maybe the whole sentence should be scaled and rotated as one...

closed 2019-01-02 by jeffeb3

billsey, 2019-01-04:
Go ahead and roll it into master, I'll grab a new master to finish up the rest of the characters. I decided to run the vertical line all the way through the '$', it shortens the array quite a bit.

Bill

Sent from my Verizon, Samsung Galaxy smartphone

jeffeb3, 2019-01-04:
I already did. Start with that, because I did the same thing. :)

billsey, 2019-01-11:
I just did a pull on the latest master. We should capture that there are additional modules needed for a build, and I'm only getting a blank page after an "npm start". It thinks it's building ok, just doesn't run right.

Bill

Sent from my Verizon, Samsung Galaxy smartphone

jeffeb3, 2019-01-11:
You have to run npm install again, I think. In the browser, there's probably an error on the console (ctrl+shft+I). I'm surprised npm didn't know about it though.

billsey, 2019-01-11:
Did an npm install, then an npm audit fix, the an npm audit fix --force to clear everything up. There are a few deprecated modules though I expect we're not using any of them. Got a import error with ToggleButton from react-bootstrap so did the install. That gave me an error with enforcePolarLimits not exported. I think I might have to wipe and reinstall npm.

Bill

Sent from my Verizon, Samsung Galaxy smartphone

billsey, 2019-01-11:
I ended up reinstalling Node.js and now I get a Failed to compile error in Computer.js. EnforcePolarLimits not exported from LimitEnforcer.

Bill

Sent from my Verizon, Samsung Galaxy smartphone

billsey, 2019-01-11:
Maybe a node.js update creating the issue?

Bill

Sent from my Verizon, Samsung Galaxy smartphone

jeffeb3, 2019-01-14:
Sorry I didn't get to this until tonight. I tried cloning a fresh checkout of the sandify code, in master. I ran npm install on it. I got a little fed up with how many things it was installing, but I think that's just javascript :). 

But it worked fine for me in chrome and in firefox. There were no errors and I got no warnings or anything in the console.

I'm using these versions, and I can't easily change or upgrade them:
npm: 3.5.2
nodejs: v4.2.6