jeffeb3/sandify #65: Wipe First Flag

billsey, 2018-12-02:
I envision a checkmark on the Sandify page that tells the gcode to do a wipe of the table before starting on the drawing. Though maybe it should be a named wipe, so you can choose a circle wipe before an angular drawing...

jeffeb3, 2018-12-02:
Hmmm. I don't like this very much. Mostly because it doesn't work well with the architecture so far. Things like reversing the path won't work if you have a bunch of wiper vertices at the beginning. The wiper tab really only works to clean from the outside to the outside. It doesn't end in the center either. You'd need a circle from the outside to the inside to get to the center.

One thing I was thinking about that I haven't quite got my head around is you create a pattern, then somehow put it into a list and make another one. The machine preview would show the previous path as gray or something, with the yellow being the one you're editing. I think I could make the list of patterns dynamic, and when you saved them, they would all be concatenated.

billsey, 2018-12-02:
I pretty much think there should only be two types of start or stop parameters (0,0 and midpoint) and everything goes from 0,0 to 0,0; 0,0 to midpoint; midpoint to midpoint; or midpoint to 0,0. Midpoint should be calculated by using the table parameters before anything else is rendered. The wipe flag just does a wipe then ignores the wipe movements in order to draw the next object. It's just a shortcut to manually embedding a specific wipe in before rendering a pattern.

mrehorst, 2018-12-25:
I have a rectangular table, so I'm not sure if this comment applies to circular tables/mechanisms.  In the past I generated wipes that started at the center and spiraled outward using a small circle with the grow and spin parameters set to appropriate values and the rotation set to zero.  Of course, you can wipe from the outside to the center by using the reverse gcode.

[image]

Be careful of spiral wipes- if you always use center to outside wipe, after a few wipes the sand will be pushed to the outside edges of the table.   Using edge to center wipe will tend to pile the sand up at the center of the table.  Alternating wipes from center to outside and outside to center helps prevent both problems.

jeffeb3, 2018-12-25:
You can approximate this with a circle, choose a small initial size, grow, and lots of loops.

jeffeb3, 2020-04-05:
Closing this. I think I've captured good solutions in these other issues.

closed 2020-04-05 by jeffeb3