jeffeb3/sandify #68: Double line text font

billsey, 2018-12-27:
Instead of sleeping I've been thinking of the variations needed to the primitives we are using for the single lined font we're currently using. For instance to generate the outline for a font like this:
[image]
We might need to create and place curves that aren't in 90° increments, such as at the top left of the V, where the serifs require different curves. I suggest that replacing the current 'bulge direction' with a number between -360 and 360 to draw circular (modified by the font scale itself, so almost all will actually be ovals) curves the indicated number of degrees (<0 for counter clockwise, 0 for straight line and >0 for clockwise) from the current position to the ending coordinates. We'd have to have a mechanism to track the starting angle in degrees from vertical as well, using that to give the direction we're curving from.

jeffeb3, 2019-01-02:
I think there are two good options for this. 

1) find a way outside of the javascript to encode this stuff. This font can be described pretty well with just a ton of line segments, which is already supported.
2) Use the gcode format for an arc, which is either clockwise or counterclockwise (G2, or G3), with a circle center, and a final coordinate. This is probably close the the most complete and compact way of describing arc segments.

bobnik, 2023-10-14:
#217 implements a version of this.

closed 2023-10-14 by bobnik