jeffeb3/sandify #70: More text characters

jeffeb3, 2019-01-02:
lowercase, international currency symbols. more punctuation. More fonts.

jeffeb3, 2019-01-25:
Thanks. It sounds like your machine has one axis going in the wrong direction. Is it a rectangular machine or a polar machine?

jaygrovr, 2019-01-25:
Core xy

jeffeb3, 2019-01-25:
Are you able to home it? It sounds like one of your motors is inverted.

jaygrovr, 2019-01-25:
It all works fine.. I guess one may be running reversed.. it was tricky getting the corexy stuff running.. so maybe in all those settings it went wonky.
Jason

jeffeb3, 2019-01-25:
The X and Y should follow the right hand rule. On mine +x is to the right from me and +y is away from me. You might have one of the homing directions inverted as well as one of the motors. If flipping a motor doesn't work, then the motors might be switched.

The perfectionist in me wants to fix your machine :). But the engineer in me wants to add an option to flip Y or X in the machine settings (not just the text). I'm just not sure how often it is for people to have a reversed machine.

If you want to debug it, you could make a post in v1engineering.com forums and I can help you figure it out.

jaygrovr, 2019-01-25:
I’ll fix it on my end.. makes everything easier to visualize and make content for.Thanks
Great work.
Jason

billsey, 2019-01-25:
I've got a few more characters ready to test. Pound, Euro, Ruble, exclamation point, etc.

jeffeb3, 2020-03-06:
Bill, feel free to push it to another branch, and I can just check it out, and copy/paste it into something to get it merged. But don't worry about it if you can't. I am going to close this issue for now.

closed 2020-03-06 by jeffeb3