jeffeb3/sandify #72: Rotation direction and reversal

Mcv888, 2019-02-04:
It would be great if the direction of rotation could be specified, but even better if the direction could be reversed during the track.  I've seen some sisyphus patterns that employ this technique and they look great.  Maybe have an additional field with number of reversals specified?
[image]

jeffeb3, 2019-02-04:
Oh. That is cool.

jeffeb3, 2019-02-23:
I'm working on this, and I'll have the reversals working shortly. I just noticed you also asked for the rotation direction. That's currently possible by making the spin amount negative. I'm guessing that's not intuitive enough to be obvious, so I'm thinking about adding a button. I'm not sure yet if I want that button.

Mcv888, 2019-02-23:
Maybe amend the descriptive note to something like “spins the input shape a little bit for each copy (positive numbers spin clockwise, negatives counterclockwise)”

jeffeb3, 2019-02-23:
[image]

This was a great idea. Thanks. I think it will look great in the sand. I added a small not that the spin step can be negative. Hopefully that helps.

closed 2019-02-23 by jeffeb3