jeffeb3/sandify #73: gcode optimization

mrehorst, 2019-02-15:
I've started working on a flow chart for a post processor for optimizing gcode.  My goal is to eliminate duplicate lines in the gcode files, but more importantly, to minimize the motion around the edges of the table while still preserving the sequence of the pattern generation.  The motion around the edges is boring to watch, takes up a lot of time, and adds to wear on the mechanism.

I've been making sketches and looking at patterns for a couple days and think I have a handle on all the special cases that have to be allowed:

Stuff to keep from the source file:
1) any non duplicate, non G01 line
2) the first G01 line
3) all non-duplicate, on-table G01 lines
4) edge to table transitions
5) table to edge transitions
6) corner to corner transitions
7) edge to corner transitions
8) corner to edge transitions
9) corner to table transitions 
10) table to corner transitions
11) edge to other edge transitions

Stuff to throw away:
1) sequential duplicate lines
2) all "in between" points along an edge
3) in between points along multiple edges

Regarding transitions-  the key point is that you have to keep anything that draws on the table, and minimize the travel between the points where the ball leaves the table (a transition) and where it reenters the table (another transition).  The source file may have the ball racing back and forth on the same edge, or around 3 corners several times before it gets back onto the table.  We can throw away all the in-between stuff and just keep (or generate) a (the shortest?) path between the ball exiting and entering the table along as many edges as it takes.

I'm trying to figure out how to make this work in a single pass, reading the source gcode line by line.  I suspect it may be easier to use multiple passes, (example- one pass that eliminates duplicate lines is very easy), but we'll see.

This is going to take a while for me to finish because I am going to have to learn whatever programming language I end up choosing once I have worked out the algorithm.  I may be able to get some coding help from people at the makerspace.

jeffeb3, 2019-02-16:
It would be really really useful to do this in the javascript. Having a python, perl, basic whatever tool the user has to use later is going to really limit the utility. If you are just making this for yourself, then that's fine, but putting it in the website would make it instantly useable by everyone.

Removing duplicates is super easy, and we can do that with about 2 lines of code in the gcode.js file.

It sounds like the rest is:

1) Watch for a transition to the edge.
2) scout ahead for the transition back from the edge.
3) replace all of that with the shortest path from transitions.

The challenges will be: a) detecting transitions, b) the data structures and managing the points safely and c) creating the shortest path (which is always harder than I think).

mrehorst, 2019-02-16:
Since there are only two paths to check for any two points on the edges of the table, I could just brute force it and calculate both path lengths and then use the shorter of the two.

Detecting the transitions isn't too hard- the Xmin, Xmax, Ymin, and Ymax are all stored in the file.  All points along edges will have at least one of those values for X or Y.  On-table point coordinates never have any of those values for X or Y.  If you compare the new input line to the last input line, you can capture all the transitions.

Keeping track of things will be the hardest thing, I think.

jeffeb3, 2019-02-16:
Is there gcode from another source you want to filter?

Putting this filter in sandify would give you access to all of the state os sandify and there would be no additional steps. Check out this file:

https://github.com/jeffeb3/sandify/blob/master/src/machine/LimitEnforcer.js

(Start at the bottom)

The trick with c) is that if you're out the xmin edge nd you need to some back through the ymax edge, It's easy. Go to xmim, ymax, then the exit point. But making that generic enough for all combinations of enter and exit edges gets tough.

mrehorst, 2019-02-17:
I just want to edit sandify gcode...

It's going to take me a while to understand that file...

jeffeb3, 2019-02-17:
I can look at this (at some point). I don't want to step on your toes though. I know that file is a mess, and if you're not familiar with javascript (I wasn't when I wrote it) it can be a real uphill climb. I'm assuming something like this should be easier than most of the file to read:

```
// Manipulates the points to make them all in bounds, while doing the least amount of damage to the
// desired shape.
function enforceRectLimits(vertices, size_x, size_y) {
  var cleanVertices = []
  var previous = null;

  for (var next=0; next<vertices.length; next++) {
    var vertex = vertices[next];
    if (previous) {
      var line = clipLine(previous, vertex, size_x, size_y);
      for (var pt=0; pt<line.length; pt++) {
        if (line[pt] !== previous) {
          cleanVertices.push(line[pt]);
        }
      }
    } else {
      cleanVertices.push(nearestVertex(vertex, size_x, size_y));
    }
    previous = vertex;
  }

  // // Just for sanity, and cases that I haven't thought of, clean this list again.
  var cleanerVertices = []
  for (var i=0; i<cleanVertices.length; i++) {
    cleanerVertices.push(nearestVertex(cleanVertices[i], size_x, size_y));
  }

  return cleanerVertices;
}
```

If you want me to look at it, posting a file with some of the error modes would help me shortcut a lot of the testing. I can find a test case for the duplicate vertices, but something that wraps around the wrong way, or moves too much along one edge would be really helpful.

jeffeb3, 2019-02-24:
I got rid of the duplicate vertices (I think). It shouldn't have any if they are within 0.001, which is the resolution of the filetype, not really a resonable resolution. But I think the actual duplicates were what was bothering you, right?

So if you're drawing a star wipe or something, and it's growing so large that it's all out of the area except for the lower left corner, and it's still got some work to do there. Instead of driving around the entire outside, it would just do little circles in that corner to finish it up? Is that right?

jeffeb3, 2019-02-24:
Oh man, this is harder than I thought. I made this same oversimplification when I started the limits enforcer. The gcode/vertices don't describe points, they describe line segments.

Both points can be on the perimeter, but part of the line segment could be running through the work area, so that needs to be found and accounted for as well.

I think I might be pretty close but I'm looking at this, and along with the other limits enforcing stuff, this is just a real mess, and there are so many if...else.. and loops that I really should write this whole thing from scratch with this in mind. I also should start with some unit tests since my system level testing is taking quite a lot of time and it's not very reliable. I'm going to give this some thought, but I think I might be convinced that the whole vertex system I have is totally wrong, and I should rewrite it from scratch...

mrehorst, 2019-02-25:
Yeah, that's what I have found, too.  I keep working on a flow chart and I keep restarting it to account for new possiblities that come to mind.  

Each G01 command represents a segment between the last G01 and the current one.  You can have a large square pattern that sends the ball from one edge to another, drawing a line across the table which will be two sequential transitions- edge to table, and table to edge- you can't throw either away.

There are a lot of possibilities to account for...

mrehorst, 2019-05-18:
After a bit of a break, I've been working on the optimization problem again.  I made a test pattern file then[ imported the text into a spreadsheet](https://docs.google.com/spreadsheets/d/1Qe_IDCPLNTesthftocyc290CnZBB8wUwoU0OmIOQfoM/edit?usp=sharing).  

I did some manipulation of the G01 commands to isolate the X and Y ordinate values (columns B and C), then calculated Cohen-Sutherland outcodes (columns H-K) for each value in the file, and decimal value of the CS outcodes in column L.  In column M, I colored the cells based on their point locations- green for on the table, yellow for the edges, and blue for the corners.  Finally, in column N, I assigned a "keep" value to the points that need to be kept.

The decision to keep a point or not is made by looking at the decimal CS outcode for the point- 0 means it is on the table and should be kept.  If it is non-zero I look at the outcodes for the points immediately before and after the current one to see if it is a transition from the table to an edge or edge to table, and keep either of those. A change from a 0 to and nonzero outcode indicates a table-to-edge (or corner) transition, and vice versa.

I haven't figured out how to create the path to follow to go from one edge to a different edge.  Simply keeping the corner points already in the file doesn't do the job.  Notice that at the end of the test pattern file (after row 9625) the ball will spend a lot of time racing around the perimeter of the table.  If you simply keep the corners, you'll end up keeping all that stuff.  You have to look at which edge (or corner) the ball exits the table and which edge (or corner) it reenters (if it does), ignoring any racing around the edges in between, then select the path to connect the two.  The shortest path will have either 1 or 2 corners, never 3 or more.  The shortest path, clockwise or counterclockwise, will be less than 1/2 the perimeter of the table.

This test pattern was made using a heart shape, so there aren't any straight line transitions that go from one edge to a different edge without any on-table points in between.  I'm going to try to apply the same logic to a pattern based on a polygon so there will be straight lines that go from one edge to a different edge across the table to see how I need to tweak the logic to make it work.  

I found a small bug in the gcode file generator- in line 12 of the spreadsheet the test pattern file says "Selected Shape: none".  I used the heart pattern...

[sandify test 400x500.txt](https://github.com/jeffeb3/sandify/files/3194586/sandify.test.400x500.txt)

mrehorst, 2019-10-29:
I have written a first crack at reducing time wasted at the edges of the table in the form of a post-processing Perl script.  It isn't a full optimization yet, but it eliminates all back and forth motion along each edge of the table which is about 90% of the pattern optimization problem.

In the first of four passes over the pattern file, the program locates the values for MinX, MaxX, MinY, and MaxY, then completes four passes with each one stripping out excess commands along the edges of the table.  Along the way it generates three intermediate files and deletes them at the end leaving just the untouched original file and the trimmed version. 

What it doesn't do (yet) is choose the shortest path between a table-to-edge and the next edge-to-table transition.  That's coming next...

I'll post a link to the code later this evening after I've tested a few patterns on my table to verify that it doesn't do anything weird (visually scanning the input and output files looks 100% good).

It turns out the algorithm is pretty simple- read three lines from the pattern file and write the second line to the output file unless all three lines contain the MinX value (or MaxX or MinY or MaxY).  It will write single and double instances of those lines, but as soon as there is a block of 3 or more lines containing the target string, only the first and last get written to the output file.

In my tests the pattern files often shrink by almost 50%.  

Here is an example- [heartfuzz1.gcode](https://drive.google.com/open?id=12OFV8rZCFbg0wNBijdf7iCD6ovYSMc6-) generated by sandify and [perledit.gcode](https://drive.google.com/open?id=199hJIvNtHpbGAMSxOw0t_N6mXDQ4RYRH), the output of my post processor.

heartfuzz1.gcode is 16,805 lines vs perledit.gcode is 8,859 lines

heartfuzz1.gcode contains 2,519 instances of MinX, 2,589 instances of MaxX, 1,145 instances of MinY, and 2,716 instances of MaxY.

perledit.gcode contains 244 instances of MinX, 246 instances of MaxX, 310 instances of MinY, and 212 instances of MaxY.

mrehorst, 2019-10-29:
Oops- didn't mean to close the issue...

mrehorst, 2019-10-30:
I've tested a few trimmed files on my table and they work perfectly.  [Here is a link to the Perl program that does the trimming. ](https://drive.google.com/open?id=1-KN-KL7aqnljWHJRP20Ux2-P--oc_QDH)  Start the program by typing "perl sandify_trim.pl"  Please try it out and let me know what you think.

I am not a Perl programmer, and this is the first program I've written in any language in at least 10 years, so go easy on me.  I wrote the program in 4 days after studying a couple Perl books and using a few google searches to find out how to do specific things, and lots of small test files.  I am sure that an even mildly-seasoned ( salt, no pepper) Perl programmer can make it better, but it's not done yet and I intend to do some of that improvement as I learn more. 

Now that the paths along the edges are reduced to the minimum, there will be a lot of corner to corner movements/coordinates in successive lines of the pattern file.  That will make the next level of optimization easier.

mrehorst, 2019-11-02:
I have completed the optimizations!
The final file is called [sandify_trim_all.pl](https://drive.google.com/open?id=1SbMMQHo6ZNLFCecdz5Lckx143fhuzlRO)
I have tested it with numerous test files and sandify patterns and checked their behavior on my sand table and it seems to be working perfectly.

Using it is very easy- start the program, type in the name of the pattern file, and get the output in a file called "clean.gcode".

The first half of the program trims out short segments and back-and-forth motion along each edge.  
 
The second half looks for sequences of statements containing corner coordinates.  Single corners are written to the output file as-is.  Two corner sequences are checked to ensure they are the shorter path through two corners and if they aren't, replaces the original two corners with the other two corners.  Three corner sequences are replaced by a single statement with coordinates of the other corner.  Finally, four corner sequences are simply cut out because they start and end on the same edge of the table.

The code is really crude- there's no range testing on variables or error trapping.  Assumptions include  the header line containing the definitions of MinX, MaxX, etc. will remain in its current form and in a location near the top of the pattern files, and that the table is rectangular.

At some point I'll probably adjust the I/O so that you'll specify the pattern and output file names in the command line.  I'll look into translating the whole thing to Java Script so it can be incorporated into the pattern generating program.

One thing I have noticed is that edges of the patterns look a little ragged after running them on the table because there's no longer any wasted motion along the edges to smooth them out.  I may add a single loop around the perimeter of the table at the end of the trimmed pattern file to clean up the edges.

mrehorst, 2019-11-04:
Further testing finds that the optimizer sometimes changes the pattern startup location if you use the "force origin" option when you save the gcode files.  It doesn't affect the pattern itself, only the movement of the ball along the edges of the table.

If I understand its correct use, the "force origin" option is used to indicate the home position of the table, so that the ball will be starting from that point assuming the table is homed before the start of each pattern.

jeffeb3, 2019-11-06:
It's that, but also used to string patterns together. If you always end or start at the lower left corner, you can connect any two patterns as long as one goes in and one goes out.

mrehorst, 2019-11-06:
Hmmm.  Maybe I'll modify the optimizer to just write out the first 3 or 4 G01 command lines in the file before it starts testing for corner to corner sequences.  That should leave the force-origin stuff alone if it's used, and will have minimal impact if it isn't.

I have modified my development version of the optimizer to add statements to the top and bottom of the trimmed output file, indicating that the file has been trimmed.  That can be useful in a file that is composed of multiple patterns.  That hasn't been posted to google drive yet.

I haven't tried to run the optimizer on a file consisting of multiple pattern files strung together.  I've been running it on the individual files and then stringing them together.  If you change the pattern dimensions from one pattern to another, running the optimizer on a composite pattern file will be a problem because it will assume the MinX, etc. for all patterns are the same as whatever the first MinX, etc. definition is.

mrehorst, 2019-11-11:
I have added a couple small things to the program- statements that appear at the top and bottom of the trimmed files, stating that they have been trimmed.  I also renamed the program to a more appropriate "trimify.pl".  [The link to the program is here.](https://drive.google.com/open?id=1SbMMQHo6ZNLFCecdz5Lckx143fhuzlRO)

I made a couple short videos of one pattern, untrimmed, and the same file trimmed running on the table.  
[Video of the untrimmed file is here.](https://vimeo.com/372267841)  
[Video of the trimmed file is here.](https://vimeo.com/372269836)

mrehorst, 2019-11-18:
After running several more patterns untrimmed and trimmed, I found that occasionally there will be some remaining excess edge motion.  It seems that if the useless edge motion happens to go around a single corner, that corner and the useless motion to it sometimes remains in the "clean"output pattern file.  A second run of Trimify on the "clean" output file from the first run removes the remaining excess edge motion.  A third run of Trimify doesn't seem to cut out anything, at least in the patterns I've looked at so far, so I think two passes will ensure complete clean up the pattern files.

I'm going to tweak a version of Trimify to make two full passes over the original pattern file.  The resulting output should be (I hope) completely free of excess edge motion.

mrehorst, 2019-12-04:
I have finished the new version of the software to trim out edge and corner to corner motion.  It is now called trimify2x.pl.  I simply copied the program and appended the copy to the end to make it run twice.  There were a few minor tweaks to input and output file names.  I modified the user i/o so that if you enter a filename like "starz.gcode" the output file automatically will be "starz_clean.gcode".

[Get trimify2x.pl here.](https://drive.google.com/open?id=1SAtsusQb1qQqGLDfiAA2qmPR3LpHocyb)

New example- here's the pattern straight out of sandify- it takes 2 hours to finish on my table:
https://vimeo.com/376736096

and here it is after running trimify2x on the pattern file- it now takes 57 minutes to complete:
https://vimeo.com/376548917

Here's that specific pattern's header so you can recreate the pattern:
; Created by Sandify
; 
;   https://sandify.org
; 
;   Sandify Version: 0.1.4
; 
;   Machine Type: Rectangular
;     MinX: 5 MaxX: 710 MinY: 5 MaxY: 1600
;   Content Type: Shapes
;     Starting Size: 300
;     Offset: X: 51 Y: -1000
;     Selected Shape: Circle
;       Circle Lobes: 1.47
;     Number of Loops: 244
;     Spin: true
;       Spin Value: -0.6
;       Spin Switchbacks: 0
;     Grow: true
;       Grow Value: 5
;     Track: true
;       Track Count: -34
;       Track Size: 880.45
;       Track Grow: false
;   Path Reversed: false

[image]

mrehorst, 2020-03-29:
I found a problem in the corner-to-corner sequencing part of the program. Trimify2x assumes that corners come in specific sequences like A-B-C, or A-B, etc., but in some pattern files, the corners can show up in sequences like A-B-A-B-B-A-C and the program doesn't handle that properly.  

You can see it in the time lapse videos here- the error shows up in the last few seconds.  First, the untrimmed file running on the table:
[https://vimeo.com/381845996](url)

Now the trimmed version- it all looks perfect until the last few seconds where you can see the ball drawing a few straight lines that go from the left edge to the bottom edge of the table:
[https://vimeo.com/381844758](url)

I compared the two files and found a strange corner sequence near the end of the trimmed file.  This seems to be a pretty rare problem- it's the first time I've seen it in any of the files I have processed using trimify2x.  I suspect that once the corner sequencing is corrected, I won't need to run the program for a second pass.

I'm going to rewrite the corner sequencing part of the program.  I think the best solution is to simply ignore the corner sequences and synthesize new ones that connect the preceding table-to-edge command with the following edge-to-table command by the shortest path, whether that means 0, 1, or 2 corners.  That should work for any and all corner-corner sequences.

jeffeb3, 2020-04-17:
I think Bob has this working. It isn't exactly the same. The Fuzzy Hear ended up being 8979 vertices. I am not sure if the extra vertices would matter.

[image]

I don't know how many vertices this is coming from your perl script. I suppose I can paste it here, and you could run it through to see what it might remove.

[image]

Here is the gcode from that second pattern. I don't have your start/stop gcode in there.
https://drive.google.com/file/d/119sAA-Zq5jWgabZ8GOhHbKAiIt9byDd3/view?usp=sharing

jeffeb3, 2020-04-17:
These changes are not on sandify.org yet. They can only be ran locally.

jeffeb3, 2020-04-17:
I just ran it through myself, and it resulted in the same file (secondExample.gcode). So I think that's a good indication that Bob's changes worked.

mrehorst, 2020-04-17:
I'll try running that pattern on my table and see how it behaves.  I had no idea that anyone was working in this!

bobnik, 2020-04-17:
> I'm going to rewrite the corner sequencing part of the program. I think the best solution is to simply ignore the corner sequences and synthesize new ones that connect the preceding table-to-edge [...]

Hey Mark. You and I were working on algorithms in parallel (great minds lol). I stumbled on your post after I'd completed an early version of perimeter optimization for polar machines. Your approach to ignore the corners and then re-add an optimal perimeter path between the points is _exactly_ the approach I took.

mrehorst, 2020-04-17:
I compared the trimify2x.pl processed file, mishegoss3_clean.gcode, to the secondexample.gcode file posted above and they look identical until near the end where the secondexample file contains some corner commands that my program cut out (that it probably shouldn't have cut), so I am hopeful that it is going to work properly.  We'll know soon enough- setting up the camera now...

mrehorst, 2020-04-17:
Looks perfect!  No end-of-the-pattern lines from the left to the bottom edges.
I'll put up a timelapse shortly.

Here's a frame from the end of the patten:
[image]

Are you cleaning up the files in a single pass?

mrehorst, 2020-04-17:
Here it is: [mishegoss3 test video](https://vimeo.com/408869150)

jeffeb3, 2020-04-17:
Very cool. Thank you both. I know this was a lot of work, and it isn't flashy, but this is like the ninja of fixes. We can just put this into the next release, and people will just get less frustrated (including us).

> Are you cleaning up the files in a single pass?

It happens seamlessly to the user, but there are multiple steps, and the code loops through the vertices many times, for different things. But in my testing, even with the developer build it has been "real time". At least an update per second, usually many more.

If you'd like to see the implementation, it is [here](https://github.com/jeffeb3/sandify/blob/master/src/features/machine/machine.js#L4). Starting with something similar to the old enforceLimits, then cleanVertices, then optimizePerimeter.

That PR (well, #124) has been merged to master, but I have to make a release to push it to sandify.org. I will wait a week or two (I just made a release) and we'll probably have a few more new features to go with it.

closed 2020-04-17 by jeffeb3