jeffeb3/sandify #74: Hitting Enter Reloads Page

jexoteric, 2019-02-20:
When entering in numbers to any of the fields, and by habit hitting enter, the page reloads erasing all current inputs for everything except the machine size fields.

Using Firefox 65.0.1

jeffeb3, 2019-02-20:
Agreed. This shouldn't happen. I'm sure I can disable it.

jexoteric, 2019-02-20:
More detail: The modifier number fields under Loops, Grow, & Spin exhibit this behavior, but the fields for Track Size, Track Length, & Grow Step do not do this. 

Also duplicated on a different computer running Chrome.

jeffeb3, 2019-02-23:
I think I've fixed this. If not, then let me know and let me know which browser and OS you're using so I can try to replicate it.

closed 2019-02-23 by jeffeb3

jexoteric, 2019-02-24:
This does indeed appear to be fixed. Thank you!