jeffeb3/sandify #75: UI Improvement

jexoteric, 2019-02-20:
A quick UI improvement would be to add chevrons to each box bar when it is collapsed to give the user a hint that it expands down. A second option that may be easier to implement would be the horizontal lines used to show menus on mobile UIs.

[image]

jeffeb3, 2019-02-20:
I will see what React Bootstrap can do. I need to find something I like. I think it is a little tricky.

jeffeb3, 2019-02-23:
I thought about this some more, and the fact that it expands isn't as important as the fact that they are buttons. When they click on them, they will expand, and if it's not expanded, then it won't affect the pattern.

The one exception is the machine settings. I can't remember why I though I should hide the settings on the second click. I might rethink that.

jexoteric, 2019-02-24:
Other than experimenting, how does a new users know easily which options are active and having an effect? I see the intent is anything that is "open"/expanded is active, but it can also just appear that anything open has a blue button (which is not clear that it means it is active). 

The chevrons or menu icons would help indicate that these are indeed buttons and can be clicked for options based on commonality with current web and mobile apps.

jeffeb3, 2019-02-24:
When you click on the Circle button, the drawing changes to a circle and the preview shows a circle. That's the kind of reinforcement I want. What's wrong with learning through experimenting?

I imagine the hurdle is just convincing people to click on the button, or to start experimenting.

jeffeb3, 2020-04-05:
The UI has changed enough that I don't think this applies is 0.2.0. I am closing, but feel free to comment if you don't think this has been addressed well enough.

closed 2020-04-05 by jeffeb3