jeffeb3/sandify #76: Basic Instructions

jexoteric, 2019-02-20:
Basic instructions or pointer tips are needed to help those just getting started understand what each variable does in order to create desired effects.

As an add-on to this, a basic getting started tutorial that goes through creating several patterns would be helpful. The faster this program is to learn, the more awesome designs people will create and share.

Background: I just finished putting together my machine and really wanted to see it working, but got flustered in trying to quickly create a pattern that would look interesting and fill the entire area.

jeffeb3, 2019-02-23:
So, I started this (I actually wrote a bunch of it up) and I have a few problems with it:
1) I would rather the interface be intuitive, and spend energy on making the flow of the website better.
2) I am worried about losing users that don't understand it on the first try, but I also worry about losing users with incomplete or confusing documentation. I am not convinced I can write documentation that will make it better. It's a very visual interface, and describing it is actually pretty hard to do. It will also only be in English, and it will need editing every time there is a change to the software. Basically, it's hard to write helpful documentation.
3) I don't like the idea of tooltips, mostly because they don't work with touch interfaces. I don't like '?' boxes because the effort would be better spent making it intuitive.

So some things I could do that make those problems smaller:
a) I could make a short video, convert it to a gif and make it easy to see. It wouldn't describe what the features did, but it would help someone make one pattern that worked, and they could play with it from there. Things like the fact that the shapes are buttons would be more obvious because you could see the interaction in the video. A second or third video might make sense to explain a particularly confusing concept.
b) I could use some feedback on which parts are the most confusing. When I first made Sandify, I called Grow "Scale" and Spin "Rotate" and I got the very helpful feedback that Grow and Spin were much simpler terms.
c) Maybe I should split some of the features into yet another tab, to keep the landing page as simple as possible. I think the Track feature, the Offsets, and maybe even the Circle are more difficult to use.

For the most part, I want people to pick it up the way they might tap the keys when first seeing a piano. Things like "Grow Step" is purposefully ambiguous, because you can push the arrows to increase or decrease it and see what it does. I'm sure this is unachievable for 100% of the users, but I think striving for that will make it better for everyone.

I definitely, absolutely appreciate the feedback, and I hope you can help me smooth this out so the next jexoteric can get rolling faster.

jeffeb3, 2019-02-23:
Would it help if?:
 - The starting size and offsets were below the shapes
 - The polygon was selected by default
 - The Grow was selected by default

What browser and OS are you using (or were using the first time)? I think some combinations don't show the arrows in the input boxes.

jexoteric, 2019-02-24:
I use Firefox on Win10 by default. From the few times I've looked, everything appeared the same in Chrome on Win10. 

Where I struggled was in getting the spiraling out trail, which it looks like you made the default settings to fix that when I looked this morning. That helps tremendously as a starting point.

I think what would be helpful is maybe 3 quick videos, or purely visual documentation of screen shots that take the users through creating something like 3-5 designs. One could be a spiral wipe, which is just a circle shape running in a polygon from the center out, and the few others could build on that, maybe one focusing on manipulating the track (as what the track looks like is incredibly different as .1-.9 compared to integers like 5 or 7, which BTW is that the intended behavior?), and another focusing on growing and rotating a square, etc. 

I don't think it needs to be complicated, and I agree that keeping things quick and light are ideal as the project is still in rapid development, but something simple is needed to get people rolling.

jeffeb3, 2019-02-24:
> as what the track looks like is incredibly different as .1-.9 compared to integers like 5 or 7, which BTW is that the intended behavior?

The intention is to not use 5 or 7. But it's an instrument. However you play it is fine with me. Would it make more sense if I changed it so a 0.2 would show as a 2 or a 20 in the UI? Would it feel more strange to put in a 500 and expect is to act similar to 20?

> Where I struggled was in getting the spiraling out trail, which it looks like you made the default settings to fix that when I looked this morning. That helps tremendously as a starting point.

I'm glad that helped.

> I use Firefox on Win10 by default. From the few times I've looked, everything appeared the same in Chrome on Win10.

The main reason I asked is that I use the arrows to change the values by steps, and I don't usually type in values. I know some configurations don't have the arrows, which might make it harder to experiment with. If you are just typing in 5, 10, 15, that often doesn't illustrate the effect like 5.1, 5.2, 5.3, etc. Do the entry boxes have arrows for changing the values?

I'll find some time to make one-two videos. If you felt generous, would you want to make the videos? No sound needed, just a screencapture of the motions, showing the the preview window while clicking on the options. I would convert it to a .gif and put it somewhere easy to find.

jeffeb3, 2020-08-06:
We created a wiki, which is where we will document things that are a bit non-obvious, but still strive for the "pick it up and play with it" interface.

https://github.com/jeffeb3/sandify/wiki

closed 2020-08-06 by jeffeb3