jeffeb3/sandify #77: UI Improvement of Preview location

jexoteric, 2019-02-24:
Another area that could be improved is the location of the preview window. I think a floating window that always stays on screen would be ideal (at least for non-mobile users) but the place I personally spent the most time messing with settings and wanting to see the changes are all the modifier options. The UI is rather chunky (not a judgement, just the best description I could think of) so not very many options fit on my screen at once, so frequent scrolling is necessary. Between options this isn't bad, but having to go to the top of the page to see the visual output of each modification I make gets very tedious fast. 

Options could be: 

- Make the visual window float so it stays in the upper right of the webpage on the screen all the time.

- Flip the visual window and the machine settings to make the window closer to the modifications options.

Thoughts and opinions from others user experiences?

jeffeb3, 2019-02-24:
> Thoughts and opinions from others user experiences?
I'm not seeing other feedback like this from other users. If someone else would chime in, that would be appreciated.

> The UI is rather chunky (not a judgement, just the best description I could think of)
One person's clunky is another person's spacious :). 

> the place I personally spent the most time messing with settings and wanting to see the changes are all the modifier options
Me too. Things are a little bigger than when I first laid this out. It used to all fit comfortably on the same screen. I would like to move the machine settings somewhere else, or maybe make two separate areas with independent scroll bars could help. I think seeing the preview whenever you're changing options is important to understand their function.

The machine settings also used to be on the top, but I didn't like that since after you set them to your machine, you never touch them again. Seems like a waste of valuable real estate.

Another option is to keep the tabs smaller. The front tab has grown a lot, and it makes it so you have to scroll on most screens. The track stuff is more advanced, or confusing, so maybe there's a way to make that go to another tab or a sub-tab and make the whole scrolling a non-issue.

Honestly though, I have no idea how to do any of this. I don't have a good sense of what's hard or easy. CSS and layout stuff is where I'm the least comfortable. I'd much rather work on the math than the UI.

I hope you don't mind, but I'm going to edit this issue to help me sort it in my list of issues.

mrehorst, 2019-11-06:
Right now we see just the windowed pattern.  I suspect the patterns contain some interesting stuff outside the window...  What I'd like to see is some way to view the entire pattern with the rectangle representing the table overlayed.  Ideally, it would be nice to be able to move the table outline over the portion of the pattern that you want to put on the table.  

It seems the patterns are always rotated around the center of the table (except when path is on).  If the above is too difficult to do, maybe just specifying the coordinates of the center of rotation would be OK so the pattern could be shifted through the window manually, by tweaking the X and Y coordinates.

jeffeb3, 2019-11-06:
Have you tried messing with the offsets?

jeffeb3, 2020-04-05:
Sandify 0.2.0 reorganizes everything, and I think bobnik has done a good job using the space we have in a productive way. I am going to close this issue, but if you still feel like it could use an improvement, please don't hesitate to say so.

closed 2020-04-05 by jeffeb3