jeffeb3/sandify #79: Problem with polar table

Netzgemuese, 2019-08-11:
Hello there,
I´m recently working on a round sandtable. So something like the sisyphus table, but its running on ramps 1.4 with gcodes. 
The problem is now that when i export my designs as a gcode it is in cartesian system (X and Y) and not in polar (angle and radius) like i expected. So i already searced for a way to transform the cartesian coordinates to polar ones and i  actually found one:
bdring build a polar pen machine and wrote a python script to do exactly what i want.
Here the link to how it works: http://www.buildlog.net/blog/2017/08/polar-pen-machine-kinematics/
and here where to download the script:http://www.buildlog.net/blog/2017/08/a-polar-pre-processor-for-the-pen-bot-written-in-python/

But with this solution there is another problem as this script espects the center/start position to be in the middle of the table. Sandify althogh has the center/start position in the lower left corner. And for that problem i didnt found any solutions jet.

Now i hope that someone could help me. Maby jeffeb3 could implement either the possibility to download directly the polar coordinate-gcode or to set the center point realy in the middle.

Thanks!

jeffeb3, 2019-08-11:
Interesting. Is this table something others will use or is it just your design? Is there somewhere I can read more about it and it's construction?

What if you tried setting the min size to -250 and max to 250? Will it be centered on zero then?

Netzgemuese, 2019-08-12:
That was a great idea! Now ive got it centered and everything is working fine.
Thank You.
And i did not follow any instructions building my table. I just tried to replicate the sisyphus from photos and videos but there are other designs similar to mine:
https://www.youtube.com/watch?v=-xGQmm7FXD4
https://www.youtube.com/watch?v=X7liebJP0JM
I think they just coded their own designs but why just dont use your existing tool?
It is a bit laborious though.
But now it works so thank you again for you idea!

closed 2019-08-12 by Netzgemuese

jeffeb3, 2019-08-12:
NP. Happy to help. I guess the reason I was asking is I am wondering if I can support your gcode straight from the tool. That has to be balanced with the fact that there are users who will not know what your machine is and consequently, any decision is possibly the wrong one. If it's just your machine, then your workaround with the script is probably the best direction forward.

gspitman, 2021-03-21:
Hey Jeffe! I was looking at building a Polar version of the ZenXY, I guess it would be the ZenPolar...

I think this option to export Polar coordinates in Sandify would be a neat option for other folks with round ideas.
Have you seen this printer? https://www.youtube.com/watch?v=OoYPUCgJ-0E maybe their firmware or slicer would be a good place to start?

jeffeb3, 2021-03-21:
We have a good solution for making polar coordinates in the thr files. You just need to convert them to angles. Depending on your mechanism, it can be easy or hard. 

I have seen that printer. AFAIK, it is completely closed source.

Netzgemuese, 2021-03-24:
Hey, got a mail that there are new comments on this topic. So i will show my point of view.
This workaround was pretty inconvenient from the beginning. Unfortunatly some Results were not good. And i mean they did not work. I tried to search what went wrong. Finally i gave up.
So there are other people, wo want to build a polar version of the zenXY. Why dont implement it in sandify? 
You said that you already have a good solution for the .thr files. I dont know, but i can imagine that it is not too hart to convert it into gcode?
Thanks!

jeffeb3, 2021-03-24:
Do you have a machine that works, but is only missing the thr to gcode conversion? There are a lot of users of sandify. I don't want to add code and possibly break or confuse people when the only machines are made of vapor.

Sorry it wasn't working for you. I wish you would have asked about it in the forums. I would happily help you find a solution. I can easily write a script that will convert thr to gcode that you can use offline, or in an online editor. Once we have that sorted, if someone made plans for an open source table, I would be more inclined to add the output to sandify. 

I am willing to change my mind. But you are not the only user I have to think about. And dont forget, I do this for fun.