jeffeb3/sandify #84: Need help generating code off sandify.org

Exo5139, 2019-11-19:
Once I set up the track how I like with shapes and modifiers and adjust my machine settings, I have a Sisyphus table, I am having trouble actually getting the code to generate.

After I click the generate code button a window comes up called code parameters.  It askes for program star and end code **1)What are these supposed to be? 0,0 or 0,1?**

Regardless of what I enter in here once I click the button to generate the thetarho code nothing happens,  **2) What am I doing wrong here? Or, where is it saving the files to?**

jeffeb3, 2019-11-19:
1) You can leave them empty. They are really more applicable to gcode, where there are speeds and homing procedures at the start and end of the gcode.
2) It should open a download dialog as though you were downloading a file. But it also has to build that file in memory first (which is usually quick, and small enough). The Theta Rho code can get quite large though, and IIRC, some browsers won't work with larger files. Try something much simpler first, like with 1 loop, and see if it's actually working. Then, if that works, you can try another browser. If another browser works, let me know so I can try to fix it (at some point).

Exo5139, 2019-11-19:
Thanks for the quick reply.  Seems like Microsoft Edge just wont open the download dialog from the sandify site.  Switched over to Chrome and works quickly for short or long tracks, from some quick testing.

jeffeb3, 2019-11-19:
OK. Thanks for reporting. I will see if I can get access to an edge browser someday to find a fix. I run Linux on all my machines at home.

billsey, 2019-11-20:
Just checked with the new, improved Edge browser that came out in the big update last week and can say it's not improved enough. It still doesn't work. No really useful errors, just couldn't download.

Bill

Sent from my Verizon, Samsung Galaxy smartphone

jeffeb3, 2019-11-20:
Does it have any "console"? In Firefox, it's in the developer tools. There might be a useful error there.

billsey, 2019-11-24:
It looks like there's some sort of developer mode in Edge but I haven't had a chance to check it out.

Bill

Sent from my Verizon, Samsung Galaxy smartphone

jeffeb3, 2020-03-06:
@billsey Can you confirm this works with Edge? I tested it here, but I would love some more eyes on it. I'm going to close it for now.

closed 2020-03-06 by jeffeb3