jeffeb3/sandify #99: Rotate text?

V1EngineeringInc, 2020-03-16:
I would love to be able to rotate text. The cursive looks great. I have always wanted to put names in front of each chair. Currently we can move it around in X and Y, if we could rotate it as a whole word, or even draw in on a circle of variable size that would allow for that pretty easy, just do each name individually and then cut and paste them together in a text editor (for now)... :wink:

jeffeb3, 2020-03-16:
Great idea. I think the next thing I am going to do is make text it's own tab. I don't think the rotate/grow makes as much sense as some of the other things I could do with it.

Allted, 2020-03-16:
Yeah, I did think text was broken for a while until I realized grow and rotate had to be off. Felt kinda silly when I realized this.

P.S. I did take a few minutes and try to figure out how the heck one might rotate the coordinates but I think python is going to take me some time to figure out.

So I made my text near an edge so I could manually get the ball back home to do multiple words. I had two ideas for this, either end a word and go to the nearest edge and then follow an edge back home (so the next word doesn't drag the ball through words.....) or even better would be to make the words start in reverse and then automatically do it the right way to get back to the edge it started at. Make sense? Cursive looks so good when written the right way in the table, it would be cool to connect them at a single edge and still have them look good, just brainstorming.

The youngsters that are hanging out over here loved watching it during breakfast drawing there names! Thanks for the simple software, only took me a few minutes to whip up some gcode to keep them busy.

jeffeb3, 2020-03-16:
> So I made my text near an edge so I could manually get the ball back home to do multiple words. I had two ideas for this, either end a word and go to the nearest edge and then follow an edge back [...]

No, it doesn't make sense...

Allted, 2020-03-16:
I'll try to take some pics or a vid later. You know how good I am with words...

jeffeb3, 2020-03-16:
Transforms are hard...

jeffeb3, 2020-03-16:
ideally, the text tab, instead of the text shape, would have similar features to this tool:

https://www.reddit.com/r/SisyphusIndustries/comments/f2h3gj/sisyphus_text_generator_2_straightcircularspiral/

This is where I got the fonts from for the cursive and sans serif (with permission and lots of help from the creator).

But I hadn't considered rotating things so they are legible to the others at the table.

Allted, 2020-03-16:
Oh wow, that makes a lot of sense. I like the straight line....I did it a worse way but kinda similar, and the spiral is awesome, an awesome option would be if you could face it the other way as well (to be read by the people sitting there).

bobnik, 2020-03-18:
> Great idea. I think the next thing I am going to do is make text it's own tab. I don't think the rotate/grow makes as much sense as some of the other things I could do with it.

If each shape has its own default values that include modifiers, you might be able to leave text in the Shapes list, and just turn Rotate/Grow off for Text only. I have some ideas for how to modify the UI to accommodate this.

V1EngineeringInc, 2020-04-17:
I love it!
Thanks go to both of you and all of your hard work.

closed 2020-04-17 by V1EngineeringInc