jeffeb3/sandify #104: Font Combo Box

jeffeb3, 2020-04-04:
In Firefox, I see this message in the console when I open the dropdown to change the font on the Text shape.

@bobnik, does this make sense to you?

That drop down doesn't really fit great with the rest of the page anyway, maybe it should be replaced with a react-select?

```
Popper: CSS "margin" styles cannot be used to apply padding between the popper and its reference element or boundary. To replicate margin, use the `offset` modifier, as well as the `padding` option in the `preventOverflow` and `flip` modifiers. 
    in DropdownMenu (at DropdownOption.js:34)
    in DropdownOption (at Shape.js:59)
    in Shape (created by ConnectFunction)
    in ConnectFunction (at ShapeList.js:73)
    in ShapeList (created by ConnectFunction)
    in ConnectFunction (at InputTabs.js:22)
    in InputTabs (created by ConnectFunction)
    in ConnectFunction (at App.js:27)
    in App (at src/​index.js:7)
```

bobnik, 2020-04-04:
Doesn't make sense, but I can replicate it, and yes, react-select would
look better here in any case.

jeffeb3, 2020-04-05:
Fixed in #106

closed 2020-04-05 by jeffeb3