jeffeb3/sandify #110: Custom Shape Input

jeffeb3, 2020-04-05:
How can we allow users to make their own shapes? Maybe they want a nike swoosh or an hour glass, or whatever? Can we support a source that has a point and click interface to make a shape? Or maybe allow importing an svg or gcode file to create the shape?

Originally in #57

billsey, 2020-04-06:
Bezier curves, select the two ends then grab and move a point on the line they form?

Bill

Sent from my Verizon, Samsung Galaxy smartphone

msurguy, 2020-04-07:
I would love to see if my Rad Lines tool's SVG output could be integrated into this software... 
Here's the tool: https://msurguy.github.io/rad-lines/ 

Here are some examples of the output: 

[image]

[image]
[image]

[image]

jeffeb3, 2020-08-06:
I think the best way to do this is with gcode/theta rho input, like #168.

Belott, 2023-10-21:
I think, this enhancement is solved with release 1.