jeffeb3/sandify #113: Import patterns aren't sticking.

jeffeb3, 2020-04-05:
When I open a .thr file, and then go to the machine tab, the vectors generated are from the shapes tab, not the most recently used tab.

bobnik, 2020-04-05:
@jeffeb3, if I understand you, I can see that's confusing, but I'm not sure the answer is to remember the last tab visited before previewing. I think that a successful .thr import should add a shape to the drop list (with configuration loaded), and change tabs to Shapes. This, of course, would require we fully support import of configuration values.

jeffeb3, 2020-04-06:
The import of thr files isn't for thr files created by sandify, it is for community generated thr files. It is mostly useful for me to draw these great thr files on my corexy machine. I also imagined it would be useful if someone wanted to scale or offset a thr file and export a thr file.

You should be able to import a thr file, then adjust your machine settings and export it without touching any of the other shapes.

I can see you saying, "We should move the import into the shape dropdown, like the wiper". Well, theoretical Bob, I think that is a good idea!

closed 2020-04-23 by jeffeb3