jeffeb3/sandify #117: Rad Lines Tool

jeffeb3, 2020-04-07:
I would love to see if my Rad Lines tool's SVG output could be integrated into this software... 
Here's the tool: https://msurguy.github.io/rad-lines/ 

Here are some examples of the output: 

[image]

[image]
[image]

[image]

_Originally posted by @msurguy in https://github.com/jeffeb3/sandify/issues/110#issuecomment-610177647_

jeffeb3, 2020-04-07:
Those are some great patterns. The website also looks great, and was really nice to use. I especially like the input equations, they can really expand the capabilities for more advanced users.

I think getting those patterns into sandify would be really great. I don't know yet the best way to do it. If I just had an input for .svgs, I think that would get pretty complicated, since the format supports so much. Exporting the vertices from Rad Lines would work, something in .json format or something.

The bigger issue is that the shapes are not connected, so they wouldn't work in sandify, which never lifts the pen. The original intention is to make patterns from one continuous line, for sand tables that can't lift the marble.

I could see incorporating some of the features from Rad Lines into the shapes and transforms here. Having an editable equation for the spin and grow would be pretty neat. I already have an issue to round over the corners of the polygon. WDYT? I don't want to step on anyone's toes.

msurguy, 2020-04-22:
@jeffeb3 couple ideas here to address your points:

- after the individual curves are generated, we can connect the start point of each curve to generate single path
- the gist of the code that is making the lines is actually pretty simple, https://github.com/msurguy/rad-lines/blob/master/src/components/Polygons.vue#L218
- I use d3 to be able to change curvature easily, but there are tools to convert svg path to vertices, for example https://github.com/pshihn/points-on-path.

As I see it, the process to make rad lines usable for sandify would be a set of JS to:
- take inputs from the user
- generate SVG paths using the code I highlighted
- run each path through https://github.com/pshihn/points-on-path to get vertices
- make the end of each path connected to the start of the next path (starting from the innermost)
- profit!

Let me know if you need more guidance

jeffeb3, 2020-04-22:
Oh!

So things like the "roundness" are a product of the svg format, not computed in your code?

1. I can't think of a way to import a random .svg that someone can upload, but I can see the benefit of using this "feature" of SVG to make the polygons, and then use those to generate vertices (as you suggest) as a source feature in sandify. Is that what you are suggesting?

2. The other interesting feature of rad lines is the fact that you are just taking inputs and calling `eval` on them with an `i`. I can think of a few places where that would be awesome. The growth and spin would be the ideal candidates. I think the only challenge there is merging the currently (relatively simple) mechanism with this evaluation approach. Let me try that, and see what I can come up with.

msurguy, 2020-04-22:
@jeffeb3 the roundness is the product of d3.line's curve type (as you can see in this example: http://bl.ocks.org/d3indepth/b6d4845973089bc1012dec1674d3aff8), but the output of d3.line is an SVG path with bezier curve calculated for you. 

Yes, it is correct, basically instead of importing arbitrary SVGs, just produce SVG paths, and run them through points-on-path lib, connect the start/end points and you'd get an array of points that you can then turn into gcode or whatever else. 

RE: 2. Yes, that was my challenge and I wanted to make it in a way that would be safe to share through URLs, so I ended up using Math.js and it's not-evil `eval` feature. It prevents executing arbitrary JS and only focuses on parsing math expressions. 

Btw, I also have another similar tool that would be perfect for square tables. It also has very simple generator under the hood, but what I'm proud of most is the "randomize" feature that comes up with good formulas 90%+ of the time: https://msurguy.github.io/flow-lines/

msurguy, 2020-05-02:
@jeffeb3 let me know if I can help in some way on this.

jeffeb3, 2020-05-02:
@msurguy, Thanks for the offer! I have my own imaginary ways of implementing this, but by all means, PRs are very welcome. I think the two features I'd like to see the most are:

1. Round corners on the polygon shape. Since that shape gets stretched and rotated, I think it would look prettt good to just round over the kernel shape. Sincce it is just a unit polygon, it should be simple to round the corners without using svgs. But any solution you're intereted in submitting is better than nothing.

2. Make a way for the growth and the spin options to be equations. I did the absolute minimum by just changing those to mathjs.evaluate and it produced really interesting patterns. I have to gracefully handle the errors that come from evaluating user equations, and I need to find a way to mix that into the UI we already have. But it seems simple enough too.

I will get to this stuff eventually. Bob has been keeping me busy lately. But if you want to play with it, you're more than welcome to it.

jeffeb3, 2020-05-08:
I have learned a lot, and I think I may have misunderstood this issue/request.

I think you're saying I should read in a SVG file, similar to how I read in a .thr or .gcode file. If the file is only one path, then that does not seem too hard (given the points-on-path) package you pointed out to create an array of vertices for each path, and then combine them by either connecting the end of one to the start of the next, or using the Graph code we already have for finding connections between the array of vertices.

So I can see that SVGO uses sax to read in the xml, and I suppose I could do the same. I was just looking at the axidraw examples, and they are already pretty complicated, with lots of groups, and options. Is there a good way to quickly get to the point of either just having a list of paths, or giving up quickly? Maybe you can point me to a javascript interface that can read in svg files and find the paths that I need?

msurguy, 2020-05-08:
No, I'm proposing generating SVG paths using `d3.line()` function (that can take curve amount) and converting those paths to vertices. I don't like the idea of reading in arbitrary SVG from file, rather I want to generate SVG paths in a way that you need (connect endpoints of each path, for example).

Maybe I'll be able to create a little demo soon. It'd be basically code from `generatePolygonData` https://github.com/msurguy/rad-lines/blob/master/src/components/Polygons.vue#L218  and code from https://github.com/msurguy/rad-lines/blob/master/src/components/ClosedPolyline.vue files, then run the output of each generated path through `points-on-path` package.

jeffeb3, 2020-05-08:
Ok, thanks for the explanation. 

The way our shapes work so far, there is a kernel shape, and then it will loop trough those kernel vertices, stretching and spinning more each loop. I opened a PR #151 where the kernel for the polygon shape starts with rounded corners.

The star shape, or the heart shape, or reauleaux shapes don't have that. I could try to put a step between the shape kernel and the loops to go through d3, svg, path-to-points. That would work for the star, but wouldn't for many of the other shapes. They aren't sampled evenly. I don't have a good place to put a "final step" to send all the vertices through d3. But that wouldn't make sense for many of the shapes.

I think this is most of the benefit of the d3 approach, so I like it better.

I still would like to replace the grow (scale) and spin (rotate) funtions with user customizable math. That may be pretty simple to do.

jeffeb3, 2020-05-14:
The master branch has rounded corners and math functions for grow and spin.

Thanks for everything, @msurguy. I am hoping to release this soon. Also in master is SVG output.

Here is a preview:
[image]

msurguy, 2020-05-15:
@jeffeb3 this is beautiful! 
Thanks for letting me know, I tried out the master branch and it worked perfectly! 

Couple cents I'd like to pitch in: 
- Don't evaluate math functions on every keystroke, rather do it on blur or on some time interval (like 1 second) since it can kill your browser or do too many calculations while trying to `eval`
- Would be cool to have Randomize button for scale and spin. I got my randomizer from @anvaka's repository here: https://github.com/anvaka/fieldplay/blob/master/src/lib/generate-equation.js , here's my take on it, with math.js functions: https://github.com/msurguy/flow-lines/blob/master/src/lib/function-generator.js

closed 2020-05-16 by jeffeb3