jeffeb3/sandify #130: Web Hypocycloid

khgrap, 2020-04-20:
Under the Web shape tool, it seems as if there's a limit to how many rotations are allowed before the pattern re-starts.  This works out to some nice symmetric and pleasing patterns if the "large circle radius" is perfectly divisible by the "small circle radius."  It would be nice if there was an option to allow the pattern to continue without it "jumping" back to the same (relative) starting point. For example, with Large set to 5 and Small set to 3, after one rotation, the chain "jumps" to the beginning of the next loop in line with the first point of the first loop.  If the pattern were to be allowed to continue, the points would converge after 3 rotations (with no growth). The linked Wolfram Hypocycloid page shows this.  Is this possible?

bobnik, 2020-04-20:
It's not possible now, but it can easily be supported. I experimented with modifications that would allow this shape (and its related shape, "Clover") to continue to rotate until they complete their pattern as you suggest. It radically changes the way these shapes draw as you change the values, but I think it's an improvement. I'll submit a PR that explores this.

Here is large circle = 190, small circle = 37 (rotates 37 times to complete the pattern).
[image]

khgrap, 2020-04-20:
Neat pattern.  This is great - thanks for exploring!

jeffeb3, 2020-04-21:
@khgrap, This update has been merged, but won't be deployed until the next release.

closed 2020-04-21 by jeffeb3

khgrap, 2020-04-21:
Sounds great. Thanks!