jeffeb3/sandify #132: Confused by Polar Gcode Output

zfrew, 2020-04-21:
I have a custom polar coordinate sand table (theta rho gantry style), and I'm really excited to use Sandify as a tool to generate patterns, but I'm having some issues with generating gcode. My table has a radius of 130.3mm, and the ball starts at the center point and ends at the perimeter. When I generate a standard 10 loop spiral that starts at the center, the first command I see is G01 X250.000 Y250.000. Given that table goes in a straight line to begin the spiral, I would expect a small movement of the linear Y axis on the order of a couple mm instead of a movement of both the X and Y axis of 250mm. Something like G01 X0 Y10 is what I would expect. 

For a polar table creating a spiral, I would also expect the rotational axis to continue increasing beyond 360. In my setup, 1 unit = 1 degree of rotation, and I expect this is what most people with polar tables are doing. In the generated gcode, neither the X nor Y axis exceeds 360 even though the shape being drawn is a continuous spiral. This admittedly makes it somewhat tricky to identify which axis is the rotational axis and which axis is the linear axis. I would expect a full 360 degree rotation for each loop of the spiral.

jeffeb3, 2020-04-21:
Gcode is for cartesian machines, and thetarho is for polar machines. Can you tell me more about your machine and firmware? I am guessing we need to support it separately.

zfrew, 2020-04-21:
Ah, it wasn't clear that the gcode wouldn't work for theta rho. I have a polar machine running off of Marlin, and currently it's configured so that G0 X10 moves the rotational axis 10 degrees and G0 Y10 moves the linear axis 10mm. The gantry I made has an issue (feature?) where the rotational and linear axis are coupled, so I need to apply a compensation to offset linear change from the rotational axis. 

I think I figured out a solution for my specific case. I can take the theta rho code generated by your program and divide the rotational axis values by 2PI to yield degrees and then multiply the linear axis by the radius of my machine to yield mm. I'll then have to apply the compensation factor to the linear axis which will be based on the linear translation of the linear axis for a given rotation of the rotational axis. That should just be a static multiplier based on the delta between subsequent rotational movements. 

I'm pretty sure that should work, and I'd be happy to let you know / share code if it does. Unfortunately, both the multiplier to convert the .thr linear axis movements to mm and the compensation factor for the linear and rotational axis coupling are specific to my configuration, so they won't be applicable to other configurations. The end result should be code that takes a .thr and turns it into working gcode on Marlin though.

jeffeb3, 2020-04-21:
> divide the rotational axis values by 2PI to yield degrees

And multiply by 360. But yeah...

Can you compensate for the linear/rotational coupling in the firmware? I suppose you are really not telling Marlin anything right now.

You will also need to think about the speeds. You can limit each axis easy enough, but you should really calculate the line segment speed in the firmware and limit the feedrate accordingly. You can do this in your post processor, but there will be limitations. If you don't do this, you will be whipping the ball around when you are far out and crawling up close.

This is all very interesting, and I will help however I can. If it ended up generalized, or with at least a few machines working this way, I would consider adding it to sandify. Right now, we would be defining it just for you.

I would really like to support this kind of machine though, so if there is a good way for me to help kick this off, I am in.

jeffeb3, 2020-04-21:
If you do go customizing the firmware for your machine, I would suggest starting with grbl or grbl_esp32. They are much simpler without all the extrusion stuff (and grbl_esp32 has a webpage built in)

zfrew, 2020-04-21:
The bias towards Marlin is mostly because I have more experience, and I have my steps / mm and sensorless homing tuned. I also think that doing all of the math at the pre-processor is going to be more portable and easy to manipulate long-term. I can actually run the gcode that Sandify outputs just fine, it just doesn't go quite where I want it to. 

Marlin is definitely overkill, I just grew up using it for building repraps and the additional features like extrusion control don't seem to be causing me any issues. I also don't plan on doing any conversions on the firmware side, and I'm hoping that it can all be done at the pre-processor. I'll cross the speed bridge once positional tuning is all set. If Marlin doesn't handle speeds properly, that should be solvable at the preprocessor although it might be a bit messy.

I could have simplified this design by using a slip ring which would have allowed me to use endstops and would have avoided the need to compensate for rotation about the center gear, but sacrificing software simplicity for mechanical simplicity seemed more fun. A version of this that works with Sandify should theoretically only require a user to input the center compensation factor (along with the already existing field for machine radius). I found that number by dividing the steps for a full motor rotation (16 microsteps * 200 steps) by the steps / mm for that axis to yield a compensation value of 46.7 for each 360 degree rotation of the rotational axis. 

I'm far from a software engineer and could certainly use help if I get stuck!

Side note: As far as I can tell, Sandify implicitly assumed I'm using a cartesian gantry when exporting gcode even though I had a polar gantry selected. That's what tripped me up, as I couldn't figure out the logic behind the gcode that was exported.

jeffeb3, 2020-04-21:
> Side note: As far as I can tell, Sandify implicitly assumed I'm using a cartesian gantry when exporting gcode even though I had a polar gantry selected. That's what tripped me up, as I couldn't figure out the logic behind the gcode that was exported.

When you select the gcode output, it outputs in cartesian. It is independent of the machine limits. I realize that isn't clear. It is sort of asking you the question twice. Once for the limits and again for the export. Maybe it should just make the assumption. I vaguely remember thinking there was a table that accepted cartesian gcode commands, but was a round table.

The way I assume your mecanics work, you have a purely rotational motor (X) and another motor that is linear, but is affected by the rotation.

X = P * Theta
Y = - Q * X + T * Radius

P, Q, T are just proportional constants.

Am I close?

zfrew, 2020-04-21:
Ah, that makes sense. The visualization remains polar even with gcode selected, so I assumed that's what it was exporting. 

Yep, looks like you've got it. X is my rotational access and Y is my linear access. Assuming I turn every line into a two item array with X as index 0 and Y as index 1: 

X = (.thr[1[0]] / PI) * 180 
Y = (.thr[1[1]] * radius) - (((.thr[1[0]] - .thr[0[0]])/360) * 46.7)
\#the Y axis finds the delta between the rotational values for two subsequent lines and multiplies the delta by the compensation factor. 

My pseudocode might be a bit confusing. I'll look around for the best way to write this, but I know if I want to do it the janky way, I can turn each line into a two object array and manipulate them that way. I could also just write a for loop and parse a bigger array with % 2 = 0 or 1. [Here's a pic of my current gantry.](https://user-images.githubusercontent.com/14201680/79915997-f9d60380-83f5-11ea-99fe-40406440df42.jpg)

jeffeb3, 2020-04-21:
That is great. It looks an awful lot like a sketch I made a long time ago:

https://forum.v1engineering.com/t/polar-table-design/8747/2?u=jeffeb3

That is exactly how I imagined it.

Those equations are real close, but it needs to be based on the theta, not the previous theta. If you go around one entire circle, you will always need to have that offset, not just until the next command. I am sure you would find that kind of stuff out with a simple square pattern or something.

closed 2020-08-06 by jeffeb3