jeffeb3/sandify #142: "Wiper" shape on polar machine travelling far along the "periphery"

ratnaweera, 2020-04-29:
Hi all,
this is more of an observation/question than an issue. I use the wiper function on my polar table, super handy! I notice it

1. travels quite far along the "periphery",
2. doubles back
3. and then starts on the next "line".

A screenshot from the sandify preview:
[image]

I might dive into the code to try and understand. But wanted to ask first: Is this the desired behaviour?

Kind regards,
Roman

ratnaweera, 2020-04-29:
I notice it is much less pronounced when the angle between the "wiper line" and the periphery is closer to 90°. This might just be an artefact at some (strange) angles. Nothing important, by any means. Just curious.

bobnik, 2020-04-29:
This is actually fixed in #124 and will be in the next release (presumably soon).

ratnaweera, 2020-04-29:
Haha, thanks. Best thing I could come up with short of actually finding where in the code this "backtracking" comes from.
I'll close this non-issue 👍

closed 2020-04-29 by ratnaweera