jeffeb3/sandify #150: Recommendations for Polar (Rho Theta) machine firmware?

akzmonster, 2020-05-08:
Hello Jeff,

I have been watching and anticipating this build for a long time and I am very near the step of firmware and setup. I am wrapping up the designs on a two stepper polar machine that uses planetaries to translate the theta and rho motion independently without needing motors attached to moving parts. It's going to be a disaster for sure.

But as I have no idea what firmware is out there for this, what would you recommend? I exported a polar file from sandify.org and noticed that the gcode is still in X and Y coordinates so I was curious if you were familiar with any firmware that already interprets those G01 (x,y) codes as (r,θ) movements? I can use a mega/ramps; uno/easydriver; pi/hat or whatever really - anything that drives two steppers.

Thanks so much!

jeffeb3, 2020-05-08:
The current output is either (theta,rho), or (x,y). The x, y is formatted for gcode and afaik, should work for almost any gcode compatible firmware. The theta rho was designed for a sisyphus table, but it can be easily consumed and used for a custom controller. 

There was just another user the other day with a similar issue: https://github.com/jeffeb3/sandify/issues/132

There is also an implementation that uses the theta rho files and a raspberry pi to do the movements:

https://forum.v1engineering.com/t/sandtrails-a-polar-sand-table/16844?u=jeffeb3

That seems like a really good way to go, and it has the potential for quite an ecosystem around it.

You are treading into new territory, but you're not alone! 

If you (or anyone else) wants a specific output format, I think I can probably manage something to work for you. So if you really want gcode where X is angle and Y is radius, I can make that happen. There will be issues with a setup like that though. Specifically, the ball will go much slower up close and much faster far away.

bobnik, 2020-08-04:
Closing this issue for now... please open a new one if you have further questions or comments, thanks.

closed 2020-08-04 by bobnik