jeffeb3/sandify #160: Further gcode optimization- edge motion speed up

mrehorst, 2020-05-25:
Wow Jeff, you've been busy!  Lots of great new patterns, and the edge motion minimizations seems to work very well.  I have been doing some work on my table- specifically, I have replaced the NEMA-23 stepper motors with 75W NEMA-17 servomotors with integrated drivers.  OMG it can fly, and the motors are quiet. [Here's video of the table mechanism running at A=10k mm/sec^2 and speed set to 2000 mm/sec](https://vimeo.com/422175403), which it actually hits in some of the longer segments in the pattern.  I haven't hit the limits yet- I need a higher current power supply to push it further, but it already can go much faster than I will ever need to draw patterns in sand- maybe I'll attach an airbrush and spray paint/ink, or LEDs to do light painting.

When the ball draws patterns in the sand, if it goes too fast, it throws the sand and covers some of the previously drawn lines, resulting in a less detailed pattern by the time it's done.  On my table I have found 500 mm/sec to be the upper limit for drawing patterns and still getting pretty good detail, because it doesn't throw the sand too much at 500 mm/sec, but that will probably vary a lot with the ball diameter and sand depth.  I run erase at 666 mm/sec and it throws the sand and leaves an almost-smooth surface on which to draw the next pattern.

The speed doesn't matter at the edges of the table.  In that case, you want to drive the ball as fast as possible to minimize time to generate a pattern.  

When I generate a pattern with sandify, I always include homing commands (G28 Y, G28 X) and a speed setting command (G01 F30000) in the custom start gcode.  It would be really nice if sandify would accept two speed values- one for drawing and one for edge motion- entered as parameters along with the table dimensions, so that you could run the edge motion much faster than the drawing motion.

Maybe time to try it with Perl again to see how/if it works. It would be a much simpler program than the edge motion optimization was.  Just add a speed command to every line in the file- lower speed for lines that are on the table and higher speed for lines that are at the edges.  It may be that with the edge motion minimization that has already been done, the effect of using two speeds wouldn't do much for the total time required to draw a pattern.

jeffeb3, 2020-05-25:
That is pretty extreme motion. It's almost like you are breaking the speed of sound, but in sand. I saw that video this morning. That is crazy. I have my ZXY current set way down and I was thinking of using smaller pancake motors instead. I typically use F2000 (33mm/s), but with heavy acceleration limits.

I don't think that is a feature we can add without making a real mess in the code. So I don't think we will implement this.

You are right that the perl script will be pretty easy, just add a speed to every point, if the point and the previous point is on the edge, use your top speed.

You could also edit the firmware to do the same thing.

mrehorst, 2020-05-30:
Hey Jeff, how do I contact Bobnik?
I've been trying out the edge motion minimization and have nit to pick....

The edge motion minimization that's being applied changes the order in which things are drawn, which drastically changes the way patterns look when they are finished.  When I was working on it I was deliberately trying to preserve the sequence so that the patterns would flow in the way that they are generated.

jeffeb3, 2020-05-30:
There is a checkmark in the machine settings. If you turn that off, it keeps the previous order, but still heavily optimizes it. The checkmark is more "destructive", but it goes even further than your optimizations did.

mrehorst, 2020-05-31:
Oh!  Good!  Thanks!

Is there any explanation of the functions that control some of the patterns anywhere?

jeffeb3, 2020-05-31:
> Is there any explanation of the functions that control some of the patterns anywhere?

No. My thought was that it would just be easy to change the values, and see what they did. I think that is mostly still true. But I am leaning towards writing up some documentation. Just need some time.

closed 2020-08-04 by jeffeb3