jeffeb3/sandify #161: second 'export' ignores shape under certain circumstances?

dholth, 2020-07-14:
Lately sandify seems to be producing the wrong output on the second export. I like to import gcode and play with shapes. But when I try to download the second shape it still gives me the first shape. In Firefox.

jeffeb3, 2020-07-15:
Interesting. I use firefox, so I can try to replicate that. What sequence of shaoes did you use to do that? Were you doing any import patterns?

dholth, 2020-07-15:
It looks like I can import any gcode, then tab over to the shapes, and the download is the imported gcode. With or without changing the parameters for the algorithmic design.

jeffeb3, 2020-07-15:
That actually makes a lot of sense. I think we can figure out a way around
that. Thanks for the details.

jeffeb3, 2020-08-04:
This should be fixed in master, since we reorganized things into layers and Import isn't a tab anymore.

closed 2020-08-06 by jeffeb3