jeffeb3/sandify #163: Pattern rotation direction- similar to issue #72

mrehorst, 2020-07-18:
I have noticed the one corner of my table tends to lose sand faster than others. I believe it is due to the counterclockwise rotation direction of patterns.  For simplest example, start with a circle, starting size 10, and make 20 loops.  The pattern will start with the ball at the center of the table and expand outward, in an increasing counterclockwise spiral. You can reverse the path, which will then make it move clockwise, but it will start at the outside and move inward. There doesn't seem to be any way to make the pattern start at the center of the table and move the ball in a clockwise spiral.

Is there some way reverse the direction that I'm missing?

jeffeb3, 2020-07-18:
You're right, I don't have any mirror or counter clockwise options. But I'm not sure that would solve your sand gathering issue.

Regardless, I think that would be a simple option to add, and it makes some sense. Thanks for the idea.

bobnik, 2023-10-14:
I missed this issue. This is supported by setting Direction on the Circle layer, though looking at it, I think the labels are opposite what they should be.

closed 2023-10-14 by bobnik