jeffeb3/sandify #179: fractional loops

jeffeb3, 2020-09-16:
When we loop, we compare to an integer. It makes sense to me to consider that to be a floating point, and then figure out how much of the pattern to draw based on the fractional distance. For example, a 6 sided polygon will produce 9 points if it loops 1.5 times.

This would make some of the other features work better, I think.

bobnik, 2021-01-08:
I believe that the current advanced feature "Draw portion of path" replicates this behavior. You can say loop = 5 times, and then specify: draw 50% of path, which only affects the last loop (so it loops 4.5 times). I think partial loops may be more intuitive than having to think of the portion of the last loop, and we could consider refactoring. I just wanted to flag that this functionality exists.

jeffeb3, 2021-01-10:
I agree. 

It would be great if 1.1 squares would also interpolate that last line segment too.

closed 2025-02-06 by jeffeb3