jeffeb3/sandify #182: Layer manipulators

jeffeb3, 2020-09-16:
There is a lot of neat things we can do with layer manipulators. Like a filter that acts on the previous layer. The transforms is kind of like that, and we could probably force the functionality into more transforms. But it might also make sense to have some layers who's only job is to manipulate the layer above.

I know Bob has started this with some limiting code, to be able to limit the previous layer to a region using the machine limits code.

I'd also like to be able to add waves, or skew/stretch patterns. Or just a general f(x,i), f(y,i) transforms.

Needs some thought. Should these "manipulators" be new layers, or should they be added to the transform area? If we make them layers, they would be easier to compound.

Should the existing transforms be converted into manipulators?

Should the transform area just be completely converted into a different kind of interface?

bobnik, 2021-01-01:
I've been working on a mask layer which works as a "manipulator". At the moment, it's a hard-wired special case that doesn't draw itself, but instead acts on the previous layer (as you describe). It's also a layer itself, which makes it easy to drag and drop it wherever you want it. This may not ultimately be how we treat this type of feature (e.g., it could attach to a layer instead of being one), but I'd like us to play around with it a bit before deciding. I can definitely see layer manipulators being made more generic.

I'm not convinced that all existing transforms should be fit into this construct, mostly because it would effectively hide a number of cool functions from casual Sandify users. At the moment, you pick a shape and start playing. As a manipulator, you'd have to be aware of its existence and then add it to your pattern.  The upside of this separation, though, would be that you could easily take your transforms and copy or move them. Maybe we could eventually move advanced transforms like track into a separate manipulator layer.

One other consideration is serialization. Ideally, we'll want to make any "big" data model changes before we support import/export of a native Sandify format, per our roadmap.

jeffeb3, 2021-01-10:
I agree. I think the original intent of this issue has been solved by adding the effects. I would like to see what else we can do in an effect. I think we have a lot of power now to do some fun skew/stretch/limiting. It would be fun to try to do some wavy distortion on something like the circle loops.

But this issue is "finished". For any new ideas, PRs are welcome.

closed 2023-10-14 by bobnik