jeffeb3/sandify #183: Track transform enhancement

texx00, 2020-09-20:
Hi,
today I got into this [video](https://www.instagram.com/p/CFPsUREDTLF/?igshid=1fzuqh8o2uuu9)

I think it can be an improvement over the track transform. As far as I know, at the moment this feature moves the shape only along a circular path. 

[image]

It could be interesting to specify some path points to be used.

I don't know however how you would like to include this.
I was thinking something like an additional option in the track transform to load an svg file with a path/polyline element or some other kind of coordinate file. These coordinates will then become the track coordinates.

If this works, some others refinements can be added like shape alignment with the path direction (for poligons for example), increasing and decreasing of the shape's size at the beginning/end of the track and path radiuses.

Another thing that could be useful for the track transform is to add a "fill loop" option in addition to the use of the track lenght property. This will make easier to achieve exactly a complete loop, which may be necessary for the "path" version.

[image]

texx00, 2020-09-20:
I just checked the other issues and this is for sure related to #181

jeffeb3, 2020-09-20:
Track is definitely confusing and hard to use. It's a pretty strong feature though.

I would like to support import svg (paths only) at some point as a whole layer. Your point makes me wonder if the imports could also go into the tracks.

The real tricky part of all this is the interface. How can we make it completelt obvious what we're doing, while not requiring linear algebra experience?

V1EngineeringInc, 2020-09-20:
> The real tricky part of all this is the interface. How can we make it completelt obvious what we're doing, while not requiring linear algebra experience?

Well, how about doubling up the interface. Not necessarily the track. Letting you use a pattern as a track. So you could use a wide spiral as your track and apply another pattern on top of it. So maybe generate a pattern, it gets saved and used as a track for the next pattern. That way could you import an svg and save it as a track?

jeffeb3, 2020-09-20:
We have another issue about wanting layers to be able to edit/transform the layers above. I can't decide if this would make it easier or harder to use. Pushing the whole track assembly in there is something I'll have to think about. I still want the spin/grow transforms to be sort of the "default" choice, so people don't feel like they need to invest a lot of energy to learn how to use sandify.

jeffeb3, 2020-09-20:
[image]
This is the kind of stuff I was working on in #181.

texx00, 2020-09-20:
I think using formulas can be really powerful for "advanced" users and confusing or even blocking for some basic users. Svg can be created in other softwares without many troubles and easily imported in my opinion... Nonetheless, I'm the kind of user that would like to play with functions so I would not regret having both options. Maybe the idea is offtopic but to help new user it may be possible to filter out some "advanced" features and keep just the simple ones (or maybe just add a mark to tell the user it is something advanced).

Still, I think the track transform is not well described and I lost more or less half an hour to get to te result of the picture above (not an impressive performance)... I did put together a branch which adds tooltip notes to the input fields. I was waiting for the other pull request to be accepted before creating a new one. I think those tips can be at least an improvement until you find a better solution.

bobnik, 2023-10-14:
@texx00 Tracks have been substantially enhanced in #275 (1.0.0), and it's quite intuitive to loop multiple shapes in an arc or spiral. Try it out when you get a chance. Closing this issue, as we don't have near-term plans to support more flexible track paths.

closed 2023-10-14 by bobnik