jeffeb3/sandify #198: pen plotting advances

jeffeb3, 2021-01-26:
I appreciate that this is mainly for sand machines, but I wonder how useful this would be with a few improvements for pen plotting. 

Just adding lifting/dropping gcode (between patterns and at limits) would be pretty cool. But what about splitting patterns up into different color pens?

https://twitter.com/v3ga/status/1354102588787011585?s=09

[image]

I am just not sure how best to add in these components, while keeping the simple creation aspect for sand tables.

ilikecheese, 2021-02-27:
I just made a pen plotter and love using Sandify! For the pen users, how about a few simple 
;Start Perimeter
and
;End Perimeter
comments around perimeter moves. I was thinking this wouldn't be too difficult to implement, and then it's a simple copy/paste for the pen users.

jeffeb3, 2021-02-28:
Thank you for sharing. I appreciate the pen users too. 

The structure of the code doesn't work well for just adding those comments. Mostly because the concept of "on a perimeter" is lost before we decide to output in gcode or thr. 

I want to have a more clever vertex type, that either includes what kind of point it is, or maybe split them into groups and have the groups have that attribute. Then we can do penup or pendown gcode snippets when changing types or just skip the perimeter moves altogether.

ilikecheese, 2021-02-28:
I'm digging into the code and trying to follow along. Would you do that when the mask clips the shape (Machine.js), or where do you see that happening?

bw4257, 2022-12-05:
I use sandify, with a wall hanging pen plotter, all the time. It's a really fun activity for the kids. Has there been any more thoughts about adding a "pen lift" feature? The ability to insert some gcode between layers, similar to the "program start code" and "program end code" when exporting could work for pen lifts, color changes, etc.

Thanks for developing this software, it's so fun and amazing!

jeffeb3, 2022-12-05:
I would love to at least be able to add gcode between patterns. I am currently working on some other stuff with Bob. But it will eventually happen, just not ready to promise when.

bobnik, 2024-11-17:
@bw4257 #302 gives you the ability to add code between layers