jeffeb3/sandify #208: Very slow, then hang

adambrower, 2021-05-22:
Nothing that mentions this issue is archived, so I must be doing something novel.

When attempting to use Sandify, every operation is excruciatingly slow. Changing even a single value (in "number of loops", e.g.) takes minutes, and duplicating a layer takes many minutes. Every operation is slo-o-o-o-o-w, and finally last night, Chrome hung with the OSX "beachball" and finally hung my Mac Pro, which is quite an achievement!

Is this a browser issue? I note that Sandify.org loads a blank page in Safari 9.1.2 and Firefox 57.0.4. I'm running Chrome  90.0.4430.212 (latest). Or a system issue? I'm running El Capitan 10.11.6.

jeffeb3, 2021-05-26:
Thanks for reaching out. 

I don't have a mac, so I don't have a way to test this in Safari, or on a mac. My computer is a 2 yo dell with an i7. But I am not coming close to maxing it out.

I do get some slow downs, but only when I use it with a very large number of points. I am often running the development version, which seems even a bit slower than the "production" version on sandify.org. 

The number of points is listed in the bottom of the image preview. Can you tell me how many points are there when it is slow? If you load the default polygon, and change the loop values, is it responsive?

I use firefox (88) in Linux, mostly. I do occasionally try it in chrome (chromium v90). I do imagine that making a very complicated pattern, with lots of points will slow down any system, and we haven't completely optimized the performance (to say the least).

adambrower, 2021-05-31:
Thanks for looking at the problem. It could be dependent on the number of points. Just tried drawing a simple 6-sided Reuleaux, then increasing the number of loops. Things were good until ~loop 15. Then each redraw began to take a little longer. At loop 50, the redraw took >2 seconds. Total points: 38350. I wonder, though, whether this could be a bandwidth matter, because sometimes performance is just fine. How many concurrent users does the githost support?

bobnik, 2021-08-05:
@adambrower It's not a concurrency issue. The drawing happens 100% in the browser. We have an open PR to improve performance that should hopefully be rolling into a release soon. I would ask you to re-test then to see if it improves.

bobnik, 2021-12-20:
@adambrower the latest Sandify release now has these performance optimizations in place. If you get a chance, can you see if it resolves your issue?

closed 2023-10-14 by bobnik