jeffeb3/sandify #212: Speeding things up while maintaining pattern quality

mrehorst, 2021-08-29:
I just finished a smaller version of my sand table and ran a few patterns and it occurred to me that if you run the table slow to maintain pattern quality, the edge motion it is also quite slow.

My machine uses a Duet 3D printer controller running reprap firmware. That firmware allows changing speed and acceleration on the fly. Maybe other controllers can't change acceleration on the fly, but they can all change speed. I was thinking that when generating a pattern in Sandify, the user could specify the pattern speed (and acceleration) and the edge speed (and acceleration). On moves that draw the pattern in the sand you could run slow with low acceleration to maximize pattern quality, and with any motion that occurs strictly at the edges of the table you could bump up the acceleration and speed to get around the edges quickly.

I don't think it would be a huge improvement, but it would certainly speed things up a bit, at least on tables that have big enough motors to handle higher speeds/accelerations.

jeffeb3, 2021-08-31:
Having generic outputs (going to gcode and to thr files) makes this harder. We don't have a concept of "on the edge" when we are creating our output. 

I could see, at some point, putting that in, for each line segment. Then the user would have to define what they wanted to happen as we entered or exited the edges (`G1 F6000`/`G1 F600`).

closed 2021-08-31 by jeffeb3

mrehorst, 2021-09-03:
I ran a test just to see what it would be like. I ran a pattern with and without the edits and found that for this particular pattern, the unedited pattern with drawing and travel speeds at 100 mm/sec took 25:42. The edited version with drawing at 100 mm/sec and edge travel at 1000 mm/sec took only 15:13.

I shot some video here. The edited pattern starts about midway through the video.

I think I'm going to write a post-processor to automate the editing of the pattern files...

bobnik, 2021-09-03:
@jeffeb3 With machine boundary clipping, we do actually know somewhat when we're on the edge. I haven't looked into feasibility, though.

mrehorst, 2021-09-03:
oops, no link to the video: https://www.youtube.com/watch?v=5ZW-likJlQ4