jeffeb3/sandify #216: Layer changes in the pattern files

mrehorst, 2021-09-23:
After trying to format some text on my table using multiple layers I've noticed that there's no indication within the pattern files of where one layer ends and the next begins. It could be useful, especially for tweaking the line position that starts text, if there was some comment placed in the pattern file to indicate when a layer change occurs.  That would allow some manual editing or even automated post-processing to do additional things at the layer changes.

jeffeb3, 2021-09-23:
I like that idea. I will take a look and see if that is easy in the code. I suspect we are calling one big "calculate" function with the whole state and getting back a bunch of vertices. But maybe it will be easier than I think.

bobnik, 2024-10-24:
Added in #302

closed 2024-10-24 by bobnik