jeffeb3/sandify #218: Saving images of patterns

mrehorst, 2022-01-30:
The preview pane of the patterns is very nice when generating patterns. If exporting the gcode would also save a copy of the preview pane as a .png or .jpg file with the same root name as the gcode file, it would be very easy to build a catalog of patterns. Then when you want to program a sequence you could select patterns from the previews.

jeffeb3, 2022-01-30:
I agree. It is a feature I have wanted for a while. 

Web browsers won't let me save more than one file per click. So there is a little trickery there. 

There is also issue #91, which was going to save an image and keep enough in the meta data to be able to load it. I still want that too.

Scuba-Ty, 2024-09-09:
Could the option be to download a zip that contains the image, thr, and maybe the Sandify file to edit in the future?