jeffeb3/sandify #219: Feature request

jaygrovr, 2022-02-04:
I know this one is rare, but the text, would be cool if you can mirror it, along with a few more fonts.  My machine is not setup right, and It would be amazing to mirror the text. 
Also great work! This tool really helps me with my sandtable!

Jason

jeffeb3, 2022-02-04:
There is a new OR for fancier text #217. But mirroring is probably better fixed on your machine. What kind of machine is it? Maybe I can help you get that right instead.

jaygrovr, 2022-02-04:
yea.. well.. i did the marlin like 2 or 4 years ago.. and now I cant find the files.. I will have to start again.  Im not even sure what board its running, but I know I have octoprint as the front end.  Its a corexy setup, pretty simple with tmc drivers.
I think my motors are getting hot too, which is annoying. I lower the current, but then they miss steps.  I think I need to overhaul it.
Jason

jeffeb3, 2022-02-04:
If you reverse the motor connector (so it goes from 1234 to 4321) that will flip and rotate 90. If you do that to both motors, I think that will just flip along one axis. You also have the ability to swap where the motors are connected. 

If you have a board with separate drv8825 or a4498 drivers, you adjust the current by the little screw. In my experience, very little current (like 200mA) is enough for these machines. But YMMV. 

If you want help, you should make a post at the V1 forums. I can help you there, but there are also a bunch of experts that would come up with good ideas.

With the shortages around, replacing the control board is a bit of a pain.

billsey, 2022-02-04:
I think the question is about drawing normal text with it mirrored below. But then I've been known to be wrong before.

Bill Seymour

closed 2026-01-09 by bobnik