jeffeb3/sandify #220: freeze bug

vincentbraillard, 2022-02-13:
Hello Jeff
Since last update of [sandify](https://sandify.org/), I Have a lot of preview freeze. this happen when I upload a new thr file or when I try to move the slider to see the path. I have try with many computer, many thr file, many web explorer but still the same!

[https://drive.google.com/file/d/1XAHuc2YL_oD9vhEKur5XW-4n9AQV_394/view?usp=sharing 1](https://drive.google.com/file/d/1XAHuc2YL_oD9vhEKur5XW-4n9AQV_394/view?usp=sharing)

I have made a very small video to let see what happen! Any idea? same to you?

thanks a lot

jeffeb3, 2022-02-13:
Thanks for reporting this.

That looks a lot like a bug I found recently. I think it has to do with the fact the preview window progress bar was computing based on a larger data length (when all three layers were present). Then the layer was deleted and that length wasn't recomputed. Then it was looking for a value in an array outside of the bounds of the array.

Can you share the thr files so I can reproduce it and make sure it is the same bug?

If it is the same bug, then it is fixed in the development branch (or will be shortly when Bobs latest branch gets merged). 

Do you have a work around? I think if you changed the first layer to a point, or didn't adjust the progress slider before deleting the layer, then I think it would be ok. I am not planning a release in the near future, so it may just stay a bug for a while.

vincentbraillard, 2022-02-13:
hi. So for sure after deleted the first standar layer, and then moving the slider, le preview freeze. But sometime only when updating a new drawing or adding a mask make the same freeze. is it the same problem? I don't know. Bellow the 2 thr file for yours tests!

[contour2022-02-08222636 (2).thr.txt](https://github.com/jeffeb3/sandify/files/8056573/contour2022-02-08222636.2.thr.txt)
[centre4.thr.txt](https://github.com/jeffeb3/sandify/files/8056574/centre4.thr.txt)

You mention trying with develop branch but how can I test it? is there a web like to the develop branch?

closed 2022-11-06 by jeffeb3