jeffeb3/sandify #221: New Sandify: loads only on Chrome, & can't add layers

adambrower, 2022-03-09:
The new Sandify is unusable for me. I'm running a Mac Pro (not Macbook) with a robust video card and mbl's of DRAM. OS is far from current, but the previous Sandify site functioned just fine under El Capitan 10.11.6. This version of the site, though, renders as a blank screen under Firefox, Safari, and even iCab and Brave. Only in Chrome does the site function at all, but I can use only a single layer, because when I add another, every subsequent click anywhere on the window triggers the "New layer" dialog box, so it isn't possible to edit parameters, or to do anything except to quit. I'm disappointed. Is there a new hardware or OS requisite I should know about?

jeffeb3, 2022-03-09:
That's a huge surprise to me. I don't know if any hardware or browser requirements. I develop and use it with Firefox. Both in the desktop and on mobile. My laptop isn't exactly new. But my phone is definitely not a power machine. 

Can you try a private window in Firefox? I wonder if there is something wrong with the stored data between versions.

You can also open the console and see if there are any errors.

jeffeb3, 2022-03-10:
> In Firefox 57

Did you really mean 57? I am not diligent about updating, but I am on firefox 97. I don't see any errors or warnings at https://sandify.org.

I don't see that error, or any others from chromium 99 (not exactly chrome, but very close).

I'd really like to help, but I don't know how to.

bobnik, 2022-03-10:
I'm running fine in Chrome 98, Firefox 98, and Safari 15.2.

Are you using security addons/plugins which modify the CSP headers to be more secure? We don't current add a CSP header at all to our site.

iwannabswiss, 2022-03-11:
I have no issues with it loading on Brave, [Version 1.36.111 Chromium: 99.0.4844.51 (Official Build) (64-bit)].

bobnik, 2022-11-06:
Closing this due to inactivity. If you continue to have trouble, please let us know.

closed 2022-11-06 by bobnik