jeffeb3/sandify #222: Issue / Request

iwannabswiss, 2022-03-11:
I have a couple of requests and a potential issue.

Let me start with the issue. I have built a CoreXY [Sand-Table](https://github.com/rdudhagra/Sand-Table) based on the design by [rdudhagra](https://github.com/rdudhagra), and the design calls for .thr files. In Sandify, if I set the machine to Rectangular machine, Max X (mm) 450 and Max Y (mm) to 450, my table, draw a design that fills the table and export it as .thr, when I play the track on my table it is nowhere near the correct size. To double-check, when I import the .thr file to Sandify, it becomes smaller than what I designed but is about the size my table appears to draw.

However, if I use the settings for a Rectangular machine and export it as Gcode, then import the file for review, it appears as the correct size and doesn't shrink. The same also applies for a Polar table; if I set the machine to Polar machine, export as a .thr file, then import, it appears to be the correct size and doesn't shrink. It just doesn't allow for .thr to be exported to the correct size for a Rectangular machine. Is it possible to export .thr files for a Rectangular machine that has the correct dimensions? Did I miss something? If not, can it be possible to export and import .thr designs for a Rectangular machine? 

My second request, could the same features for the Start and End points of a Polar table also be added to a Rectangular table?

Thanks!

jeffeb3, 2022-03-11:
I'm not sure how the files work on a rectangular table. The files are "normalized". Meaning the max radius is 1.0, no matter what size the table is. Can you post a the file that fills your whole table?

iwannabswiss, 2022-03-12:
[3_sandify_files.zip](https://github.com/jeffeb3/sandify/files/8236596/3_sandify_files.zip)

Wow!  @jeffeb3, thanks for the quick response.  Attached is a zip file, I got a warning that a .thr file isn't supported, with the three variations; two that work for the "correct machine," and one for what I'm trying to do.

sandify_star_rect_gcode_450x440mm.gcode; this one imports fine if machine is set to rectangular.
sandify_star_polar_thr_450mm.thr; this one imports fine if machine is set to polar.
sandify_star_rect_thr_450x440mm.thr; this one does not import to the correct size:

jeffeb3, 2022-03-12:
If you don't import anything, you just export something from sandify, and run it on your table. Does it draw the right size? If not, that's what I want to fix first. I was asking for a thr file (that wasn't made in sandify) that drawn the right shape on your table.

jeffeb3, 2022-03-12:
I can see this isn't working. I will have to take a closer look.

I made this pattern (just a single loop, big square):

[image]

Looking at the thr, this is what I see:
```
0.78540 0.49183
0.78086 0.48961
0.77628 0.48741
0.77166 0.48521
0.76699 0.48302
...
-5.47786 0.48232
-5.48254 0.48450
-5.48718 0.48669
-5.49177 0.48890
-5.49632 0.49111
-5.49779 0.49183
```
The thr file turns straight lines into many, many points, which is why there are more than 4 lines there.

Each line is an angle, and a radius, normalized to 1.0.

The fact that it starts and ends near 0.49 and not 0.98 tells me there is a scaling issue, for sure. My guess is that I am not calculating the radius in any reasonable way when I have rectangular limits. I am guessing that will take a while to find (and I can't work on it _right_ now).

But it needs to get fixed. I will try to carve out some time.

I don't have a good workaround, either.

On your machine, is a radius of 1.0 equal to the corners, or the sides?

iwannabswiss, 2022-03-13:
> If you don't import anything, you just export something from sandify, and run it on your table. Does it draw the right size? If not, that's what I want to fix first. I was asking for a thr file (that wasn't made in sandify) that drawn the right shape on your table.

When I just export from Sandify, using Rectangular machine settings, and run on the table, it does not draw the right size; it is much smaller, as what appears when I import back.  However, if I export something written from Sandify using Polar machine settings, it does draw the correct size.  I haven't tried a file that wasn't made on Sandify, I just got the table together, so I haven't played much with it.  I'll find a .thr file.

> On your machine, is a radius of 1.0 equal to the corners, or the sides?

If I understand correctly, using the principle of a square inscribed in a circle, a radius of 1.0 should be equal to the corners for my machine, or at least two of the corners.

jeffeb3, 2022-03-13:
You can make a the file like this:
```
0 0
628 1
```
That will make 100 circles and end up at the max distance. That makes perfect sense for a round table like sisyphus. What I can't figure out is if that ends up as a circling inside the square of your table (how do you get to the corners) or if it cuts off the sides.

iwannabswiss, 2022-03-13:
Yes, I can and did.

Upon the initial load, the png that's created makes it appear that it'll end up as a circle within the square and won't reach the corners.  I'm running it and will report back.

iwannabswiss, 2022-03-14:
Ok, so after running the track, it ends up as a circle inside of the square of the table and does not reach the corners.

PR #303 (2024-10-24): fix maxRadius calculation
Fixes #222 and #285.

I believe this is as easy as correctly calculating the maxRadius when exporting to a rectangular table. We were calculating a radius double the actual maximum. This bumps the max rho to 1 in the exported file.
closed 2024-10-24 by jeffeb3