jeffeb3/sandify #226: Bug? in exporting pattern file

mrehorst, 2022-04-09:
I generated a pattern and exported it without forcing the origin, then forced origin to LR corner and exported again without making any other changes. The second version, with the origin forced to LR corner, is missing all the settings that generated the pattern:
[image]

The files are identical after the few green lines on the right.

jeffeb3, 2022-04-10:
>  The second version, with the origin forced to LR corner, is missing all the settings that generated the pattern:

That is a bug. Good find. I bet it is having an error with the corners forced and skipping the rest of the comments. 

> I found that when I exported the first file, the pattern parameters were displayed. When I exported the second file, the machine parameters were displayed. I switched back to the pattern parameters display and exported a third time, and this time the pattern parameters were displayed. Is it supposed to work that way?

Are you saying the tab (on the right side of sandify) determined if the settings were exported correctly or not? That should not have an effect on the export. But I will test that.

mrehorst, 2022-04-10:
Yes, that's what is happening. If the "draw" tab is selected when you export the gcode, the pattern parameters will appear in the gcode file. If the "machine" tab is selected the parameters don't appear in the gcode file. If you select the "about" tab and export, you get no parameters, same as selecting the "machine" tab.

jeffeb3, 2022-04-10:
Ok. Thanks for finding that. It should be easy enough to find.

bobnik, 2023-10-13:
As of #275, Sandify no longer exports settings within gcode and thr files. Those settings can be saved as a .sdf file, which produces an exact pattern replica.

closed 2023-10-13 by bobnik