jeffeb3/sandify #234: When multiple layer, some line doesn't exist before the exporting but is show on the exported file

vincentbraillard, 2022-07-25:
Let me show you. I have done multiple layer drawing and on the preview all is ok! the lines between layer are ok:
[image]
but once I export it and import in sandify the file generated, some line is show. it is just like when ending the layer, the ball return to the end of the preview layer and then return to the end of the drawing...
[image]
this is the file:
[Clover Variation.thr.txt](https://github.com/jeffeb3/sandify/files/9184972/Clover.Variation.thr.txt)

Any idea???. this happened not only with ths drawing but with multiple layer

vincentbraillard, 2022-07-27:
Someone else has the same problem?

jeffeb3, 2022-07-28:
I am not ignoring this. I will check it out when I get a minute. I have just had a very busy couple of weeks.

vincentbraillard, 2022-07-30:
Thanks jeff.🙏

amatismiles, 2023-05-01:
I have a similar issue where the ball returns to the start of the fancy text layer after it completes the text (destroying the text it just made).  This line does not show in the preview, but it does show when I import the thr file back in.

bobnik, 2023-10-13:
#275 may have fixed this. When you get a chance, please re-export and try again, thanks.

bobnik, 2023-10-14:
Closing this issue. If you see again, please upload a .sdf file that illustrates the bug, thanks.

closed 2023-10-14 by bobnik